iceman1001/proxmark3 #168: macOS homebrew broken

n0ncetonic, 2018-02-06:
```
Last 15 lines from /Users/user/Library/Logs/Homebrew/proxmark3/02.make:
mv -f obj/inflate.Td obj/inflate.d
gcc -MT obj/inffast.o -MMD -MP -MF obj/inffast.Td  -std=c99 -D_ISOC99_SOURCE -DPRESETS -I. -I../include -I../common -I../common/polarssl -I../zlib -I../uart -I/opt/local/include -I../liblua -Wall -g -O3 -DHAVE_GUI -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED -c -o obj/inffast.o ../zlib/inffast.c
mv -f obj/inffast.Td obj/inffast.d
gcc -MT obj/inftrees.o -MMD -MP -MF obj/inftrees.Td  -std=c99 -D_ISOC99_SOURCE -DPRESETS -I. -I../include -I../common -I../common/polarssl -I../zlib -I../uart -I/opt/local/include -I../liblua -Wall -g -O3 -DHAVE_GUI -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED -c -o obj/inftrees.o ../zlib/inftrees.c
mv -f obj/inftrees.Td obj/inftrees.d
awk -f usb_cmd_h2lua.awk ../include/usb_cmd.h > lualibs/usb_cmd.lua
awk -f default_keys_dic2lua.awk default_keys.dic > lualibs/mf_default_keys.lua
g++  obj/proxmark3.o obj/uart_posix.o obj/uart_win32.o obj/ui.o obj/util.o obj/util_posix.o obj/scandir.o obj/crapto1/crapto1.o obj/crapto1/crypto1.o obj/mfkey.o obj/tea.o obj/polarssl/des.o obj/polarssl/aes.o obj/polarssl/bignum.o obj/polarssl/rsa.o obj/polarssl/sha1.o obj/polarssl/sha256.o obj/loclass/cipher.o obj/loclass/cipherutils.o obj/loclass/ikeys.o obj/loclass/hash1_brute.o obj/loclass/elite_crack.o obj/loclass/fileutils.o obj/whereami.o obj/mifarehost.o obj/parity.o obj/crc.o obj/crc16.o obj/crc64.o obj/legic_prng.o obj/iso15693tools.o obj/prng.o obj/data.o obj/graph.o obj/cmddata.o obj/lfdemod.o obj/emv/crypto_polarssl.o obj/emv/crypto.o obj/emv/emv_pk.o obj/emv/emv_pki.o obj/emv/emv_pki_priv.o obj/emv/test/cryptotest.o obj/emv/apduinfo.o obj/emv/dump.o obj/emv/tlv.o obj/emv/emv_tags.o obj/emv/dol.o obj/emv/emvcore.o obj/emv/test/crypto_test.o obj/emv/test/sda_test.o obj/emv/test/dda_test.o obj/emv/test/cda_test.o obj/emv/cmdemv.o obj/cmdanalyse.o obj/cmdhf.o obj/cmdhf14a.o obj/cmdhf14b.o obj/cmdhf15.o obj/cmdhfepa.o obj/cmdhflegic.o obj/cmdhficlass.o obj/cmdhfmf.o obj/cmdhfmfu.o obj/cmdhfmfhard.o obj/hardnested/hardnested_bruteforce.o obj/cmdhfmfdes.o obj/cmdhftopaz.o obj/cmdhffelica.o obj/cmdhw.o obj/cmdlf.o obj/cmdlfawid.o obj/cmdlfcotag.o obj/cmdlfem4x.o obj/cmdlffdx.o obj/cmdlfguard.o obj/cmdlfhid.o obj/cmdlfhitag.o obj/cmdlfio.o obj/cmdlfindala.o obj/cmdlfjablotron.o obj/cmdlfnexwatch.o obj/cmdlfnedap.o obj/cmdlfnoralsy.o obj/cmdlfpac.o obj/cmdlfparadox.o obj/cmdlfpcf7931.o obj/cmdlfpresco.o obj/cmdlfpyramid.o obj/cmdlfsecurakey.o obj/cmdlft55xx.o obj/cmdlfti.o obj/cmdlfviking.o obj/cmdlfvisa2000.o obj/cmdparser.o obj/cmdmain.o obj/pm3_binlib.o obj/scripting.o obj/cmdscript.o obj/pm3_bitlib.o obj/protocols.o obj/cmdcrc.o obj/reveng/preset.o obj/reveng/reveng.o obj/reveng/cli.o obj/reveng/bmpbit.o obj/reveng/model.o obj/reveng/poly.o obj/reveng/getopt.o obj/bucketsort.o obj/proxgui.o obj/proxguiqt.o obj/proxguiqt.moc.o obj/hardnested/hardnested_bf_core_NOSIMD.o obj/hardnested/hardnested_bitarray_core_NOSIMD.o obj/hardnested/hardnested_bf_core_MMX.o obj/hardnested/hardnested_bitarray_core_MMX.o obj/hardnested/hardnested_bf_core_SSE2.o obj/hardnested/hardnested_bitarray_core_SSE2.o obj/hardnested/hardnested_bf_core_AVX.o obj/hardnested/hardnested_bitarray_core_AVX.o obj/hardnested/hardnested_bf_core_AVX2.o obj/hardnested/hardnested_bitarray_core_AVX2.o obj/hardnested/hardnested_bf_core_AVX512.o obj/hardnested/hardnested_bitarray_core_AVX512.o obj/deflate.o obj/adler32.o obj/trees.o obj/zutil.o obj/inflate.o obj/inffast.o obj/inftrees.o -L/opt/local/lib -L/usr/local/lib -lreadline -lpthread -lm ../liblua/liblua.a -F/usr/local/Cellar/qt/5.10.0_1/lib -framework QtWidgets -F/usr/local/Cellar/qt/5.10.0_1/lib -framework QtGui -F/usr/local/Cellar/qt/5.10.0_1/lib -framework QtCore -o proxmark3
Undefined symbols for architecture x86_64:
  "_showBanner", referenced from:
      _main in proxmark3.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [proxmark3] Error 1
make: *** [client/all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/iceman1001/homebrew-proxmark3/issues
```

n0ncetonic, 2018-02-06:
FIxed and submitted pull request

iceman1001, 2018-02-06:
...well  the reason for me removing the APPLE directive was because OSX didn't show the ansi-colors.
so either the solution is to remove APPLE a bit down in the source ,  not calling showbanner at all,   or reverted to include APPLE again.

Would you mind verfing the banner becomes blue on OSX  ?

iceman1001, 2018-02-06:
Until verification that ansi-colors works on OSX,  that feature is turned off.

closed 2018-02-06 by iceman1001

n0ncetonic, 2018-02-06:
https://www.dropbox.com/s/5e43bdwvfy6q6uo/Screenshot%202018-02-06%2010.05.57.png?dl=0

Looks blue to me

iceman1001, 2018-02-06:
Great! 

Its has been reverted,  
https://github.com/iceman1001/proxmark3/commit/09b2a079ba713ca25c1b12d7726965d029f0d3d7