iceman1001/proxmark3 #170: compiling with -DWITH_LF_SAMYRUN still uses iso14443a reader mode

tuxthemadpenguin, 2018-02-07:
1) Replaced -DWITH_HF_YOUNG with -DWITH_LF_SAMYRUN in armsrc/Makefile
2) make clean && make all in root
3) flash bootloader and full image
4) Run Proxmark client
4) Put Proxmark into Standalone mode
5) Output:
```
#db# Stand-alone mode! No PC necessary.          
#db# Enabling iso14443a reader mode for [Bank: 0]...     
```
Tested and it reads HF chips and not LF chips.

Happy to close if I misunderstand the docs.

iceman1001, 2018-02-07:
...did you follow the 'not-so-clear' guideline from here? https://github.com/iceman1001/proxmark3/tree/master/armsrc/Standalone

especially the part mention briefly of  RunMod inside appmain.c?

tuxthemadpenguin, 2018-02-07:
Hmmm, I'll have to read this a little more. I understood it to be that LF_SAMYRUN was already implemented in code and it was just a matter of swapping out flags in the Makefile unless I'm implementing a new standalone mode.

Should I close this out and take it to the forums instead?

iceman1001, 2018-02-07:
the source is implemented but the switching to it is almost a matter of swapping out flags.. well. one more place than the makefile.

tuxthemadpenguin, 2018-02-08:
OK, I think I'm in over my head, the tag is in a conditional with a call to RunMod() at the end.  I can't see an obvious place, to put the other flag.

I did find another location in the makefile down on line 103 where hf_young.c needed to switch with lf_samy.c .

I then have a type error I haven't figured out.  Tried to cast the variables as ints in various places, but I don't know enough about what I'm doing and ended up with an int instead of a pointer to an int.

```
arm-none-eabi-gcc -c -I../include -I../common -Wall -Werror -pedantic -Wunused -std=c99 -DWITH_CRC -DWITH_ISO18092 -DON_DEVICE -DWITH_LF -DWITH_HITAG -DWITH_ISO15693 -DWITH_LEGICRF -DWITH_ISO14443b -DWITH_ISO14443a -DWITH_ICLASS -DWITH_FELICA -DWITH_HFSNOOP -DWITH_LF_SAMYRUN -fno-strict-aliasing -ffunction-sections -fdata-sections  -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED -I../zlib -I. -Os -mthumb-interwork -o obj/lf_samyrun.o ../armsrc/Standalone/lf_samyrun.c 
../armsrc/Standalone/lf_samyrun.c: In function 'RunMod':
../armsrc/Standalone/lf_samyrun.c:52:4: error: passing argument 2 of 'CmdHIDdemodFSK' from incompatible pointer type [-Werror]
    CmdHIDdemodFSK(1, &high[selected], &low[selected], 0);
    ^
In file included from ../armsrc/Standalone/lf_samyrun.h:17:0,
                 from ../armsrc/Standalone/lf_samyrun.c:11:
./apps.h:89:6: note: expected 'uint32_t *' but argument is of type 'int *'
 void CmdHIDdemodFSK(int findone, uint32_t *high, uint32_t *low, int ledcontrol);
      ^
../armsrc/Standalone/lf_samyrun.c:52:4: error: passing argument 3 of 'CmdHIDdemodFSK' from incompatible pointer type [-Werror]
    CmdHIDdemodFSK(1, &high[selected], &low[selected], 0);
    ^
In file included from ../armsrc/Standalone/lf_samyrun.h:17:0,
                 from ../armsrc/Standalone/lf_samyrun.c:11:
./apps.h:89:6: note: expected 'uint32_t *' but argument is of type 'int *'
 void CmdHIDdemodFSK(int findone, uint32_t *high, uint32_t *low, int ledcontrol);
      ^
cc1: all warnings being treated as errors
../common/Makefile.common:80: recipe for target 'obj/lf_samyrun.o' failed
make: *** [obj/lf_samyrun.o] Error 1

```

iceman1001, 2018-02-08:
aha..  now that is a bug,   I've pushed a fix for it.

tuxthemadpenguin, 2018-02-09:
Pulled the latest, it compiles now but is not reading the cards correctly.

If I flash back to master with LF standalone
```
test@test-VirtualBox:~/newProxmark/proxmark3/client$ sudo ./proxmark3 /dev/ttyACM0 
Prox/RFID mark3 RFID instrument          
bootrom: iceman/master/ice_v3.1.0-532-gfe34cac-dirty-unclean 2018-02-08 16:56:27
os: master/v3.0.1-51-g53814fe-dirty-suspect 2018-02-07 20:05:05
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/05/17 at 17:48:26
          
uC: AT91SAM7S512 Rev B          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 512K bytes. Used: 194588 bytes (37%). Free: 329700 bytes (63%).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          
proxmark3> lf search
NOTE: some demods output possible binary
  if it finds something that looks like a tag          
False Positives ARE possible
          

Checking for known tags:
          
HID Prox TAG ID: 10000ec9a (30285) - Format Len: 37bit - FC: 4096 - Card: 30285          

Valid HID Prox ID Found!          
#db# Stand-alone mode! No PC necessary.          
#db# Playing          
#db# 0 300000202f54          
#db# Tags can only have 44 bits. - USE lf simfsk for larger tags          
#db# Done playing          
#db# Starting recording          
#db# TAG ID: 10000ec9a (30285) - Format Len: 37bit - FC: 4096 - Card: 30285          
#db# Stopped          
#db# Recorded 1 10000ec9a          
proxmark3> exit
```

When I flash to current version the iceman fork 
```[ ARM ]
 bootrom: iceman/master/ice_v3.1.0-532-gfe34cac-dirty-unclean 2018-02-08 16:56:27
      os: iceman/master/ice_v3.1.0-532-gfe34cac-dirty-unclean 2018-02-08 16:56:29
 [ FPGA ]
 LF image built for 2s30vq100 on 2017/10/25 at 19:50:50
 HF image built for 2s30vq100 on 2017/11/10 at 19:24:16
          
 [ Hardware ]           
  --= uC: AT91SAM7S512 Rev B          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 236353 bytes (45%) Free: 287935 bytes (55%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory          

pm3 --> lf search
NOTE: some demods output possible binary
  if it finds something that looks like a tag          
False Positives ARE possible
          

Checking for known tags:
          
HID Prox TAG ID: 10000ec9a (30285) - Format Len: 37bit - FC: 4096 - Card: 30285          

Valid HID Prox ID Found!          
#db# Stand-alone mode! No PC necessary.          
#db# [+] Playing          
#db# [+] 0 1 0000ea59          
#db# [+] Done playing          
#db# [+] Exiting    
  ```
I had trouble getting it in to record mode
```
#db# [+] Done playing          
#db# [+] starting recording          
#db# Stopped          
#db# [+] recorded 0 1 00007307          
#db# [+] Playing          
#db# [+] 0 1 00007307          
#db# [+] Done playing          
#db# [+] Exiting  `

h4waii, 2018-02-21:
I'm also having similar issues. I updated the `armsc\Makefile` in both places, compiled successfully, and flashed.

Trying to record the tag to one of the banks, the tag isn't being detected.

iceman1001, 2018-02-22:
Fix for lf_proxbrute  https://github.com/iceman1001/proxmark3/commit/1c37981430f0def62c2b115f72a9d278fa4bd6d1
for the compilation errors.

tuxthemadpenguin, 2018-02-25:
@iceman1001 I'm sorry but I can't figure out how RunMod() in appmain.c needs to be modified.  Can you give me an example?  It looks like it should just be using information set as flags in the Makefile and then each c file for the respective Standalone mode has a RunMod() function.  What am I missing/not understanding?

Thanks in advance.

iceman1001, 2018-02-25:
The runmod() is not what must be changed,  its the macro's surrounding them that must be adapted.

First there is  a check if firmware is compiled with support for LF,   and HF 14a,  
then it checks which standalone mode to compile in.

I pushed my appmain.c to show how I added the HF_COLIN..

iceman1001, 2018-02-25:
@tuxthemadpenguin   and this issue?  close?

tuxthemadpenguin, 2018-02-25:
I finally understand what you mean.  Thank you.  

Should I go back and check the SAMYRUN standalone?  It used LF but the output when hooked up to the client made it look like it was not interpreting the tags right.

iceman1001, 2018-02-25:
for sure,  I would like you to check it out to see if there is a bug or not

tuxthemadpenguin, 2018-02-25:
Yes it does not seem to function to properly compared to the LF mode in the main branch/master

Proxmark main branch LF standalone 

```
#db# Starting recording          
#db# TAG ID: 2005f40d42 (1697) - Format Len: 26bit - FC: 250 - Card: 1697          
#db# Stopped          
#db# Recorded 0 2005f40d42          
#db# Playing          
#db# 0 2005f40d42        
```  

current Iceman compiled with SAMYRUN

```
#db# Stand-alone mode! No PC necessary.          
#db# [+] playing          
#db# [+] 0 1 0000de03          
#db# [+] done playing          
#db# [+] exiting          
#db# Stand-alone mode! No PC necessary.          
#db# [+] playing          
#db# [+] 0 1 0000eb0f          
#db# [+] done playing          
#db# [+] starting recording          
#db# Stopped          
#db# [+] recorded 1 112ea9 fffcc200          
#db# [+] playing          
#db# [+] 1 112ea9 fffcc200          
#db# [!] tags can only have 44 bits. - USE lf simfsk for larger tags          
#db# [+] done playing          
#db# [+] playing          
#db# [+] 0 1 0000eb0f          
#db# [+] done playing       
```   

It is coming up with odd output.  It is more difficult for me to put in to record mode once in standby.  When I do switch it to record it isn't detecting tags until I hit the button again then it claims to be playing a new tag.   

I was testing to see if it was just a display output problem(using a white scanner/cloner) but it is not playing back the same card value.

Let me know if you need more information.

iceman1001, 2018-09-05:
I pushed fixes into RRG repo.

iceman1001, 2018-09-18:
@tuxthemadpenguin   ping

closed 2019-06-24 by iceman1001