iceman1001/proxmark3 #174: hf 15

osysltd, 2018-02-14:
Hi Iceman,
Just wanted to share the outcome of working with 15a tag, might be there are some timing problems?
```
Reading memory from tag UID XX XX XX XX XX XX XX XX          
#db# [!] error, uneven octet! (extra bits!) mask 02          
iso15693 card select failed          
...#db# [!] error, uneven octet! (extra bits!) mask 02          
iso15693 card select failed          
.........#db# [!] error, uneven octet! (extra bits!) mask 02          
iso15693 card select failed          
.#db# [!] error, uneven octet! (extra bits!) mask 02          
iso15693 card select failed          
.#db# [!] error, uneven octet! (extra bits!) mask 02          
iso15693 card select failed          
#db# [!] error, uneven octet! (extra bits!) mask 02          
iso15693 card select failed          
..#db# [!] error, uneven octet! (extra bits!) mask 02          
iso15693 card select failed          
#db# [!] error, uneven octet! (extra bits!) mask 02          
iso15693 card select failed          
#db# [!] error, uneven octet! (extra bits!) mask 02          
iso15693 card select failed          
#db# [!] error, uneven octet! (extra bits!) mask 02          
iso15693 card select failed 
```

iceman1001, 2018-02-14:
Could be timing..  usually when I get that its antenna & positioning.

p.s
iso15693 doesn't have a...  ;)

ceres-c, 2018-02-14:
I have to back this issue up as I'm facing the same error. I believe this is due to recent CRC16, commit 52d69ed4eefa2df7e7c45ef39c22ee26093ec743

Command `hf 15 raw -c 26 01 00` gets no octets and
> #db# [!] error, uneven octet! (extra bits!) mask 02
pops up in terminal

If I instead run `hf 15 raw 26 01 00 f6 0a`, with the correct CRC16 calculated by official fw and not on the fly, I get the correct answer and there is no uneven octect error :)

iceman1001, 2018-02-14:
Now,  thats a bug.  Seems to be related to  **hf 15 raw**
Most likely to do with the CRC change.

iceman1001, 2018-02-14:
and I pushed a fix for it.

ceres-c, 2018-02-14:
Thanks, it's now fixed, no more complaints from `hf 15 raw -c`
Regarding 'uneven octet' i have made a PR implementing a dumb fix. I believe the error is wrongly reported when no octets are received, so checking for received octets before should be enough... Let me know, since you know pm3 way more than me

iceman1001, 2018-02-14:
@ceres-c  Great!    I merged it.  grazie.
@osysltd  Would you mind testing and see if your bug still exists?

iceman1001, 2018-02-14:
`pm3 --> hf 15 reader
 UID  : E0 07 00 00 14 7C 95 B6
 TYPE : Texas Instrument; Tag-it HF-I Plus Inlay; 64x32bit
`

With a reading distance of 10cm.   **awesome**

ceres-c, 2018-02-14:
@iceman1001 great!
I have tried to select a tag with your fw and then reading the said selected tag, but I get no response even if selection is successful. Launching raw commands via both your and standard fw gives the same result. I have no interest in this function, but I believe something's still broken.
My guess goes to the "52" sent as a flag before the read tag command. I am not 100% sure that's correct but I could be wrong and it might be that my tags are not fully compliant (even if the datasheet says so)

iceman1001, 2018-02-14:
Not following.  You tried which commands and what happend (output/trace etc) ?

ceres-c, 2018-02-14:
Trace:
```
pm3 --> hf 15 select
#db# [+] SEND          
#db# &....    26 01 00 f6 0a          
#db# [+] Iso15693InitReader Exit          
#db# ice: demod bytes 12          
Detected UID E0 04 02 00 00 A4 1F C4          
#db# [+] RECV          
#db# No error [+] crc OK          
#db# ........ 00 03 c4 1f a4 00 00 02          
#db# ..7.     04 e0 37 04          
#db# [+] SEND          
#db# "%...... 22 25 c4 1f a4 00 00 02          
#db# ....     04 e0 ea a3          
#db# [+] Iso15693InitReader Exit          
#db# ice: demod bytes 3          
Card is selected          
#db# [+] RECV          
#db# .x.      00 78 f0 
pm3 --> hf 15 read s 0
#db# [+] SEND          
#db# R ...    52 20 00 a4 d3          
#db# [+] Iso15693InitReader Exit          
#db# ice: demod bytes 0          
iso15693 card select failed          
#db# [+] RECV
```

As you can see the tag answers correctly to "select" (00 78 f0), but the second command, "read", fails with no response. I believe this is due to the flags sent with the read command, 52.
Not sure about this one, but it's the only reason I can find...

PS Got same behaviour on both a NXP and EM tag.

iceman1001, 2018-02-14:
...hf 15 select..  not sure it works very well.  Looking at your trace, it definitely looks broken.
its the 15 command set where you can "select" a tag,   (instead of using 15 info u etc)  and run commands against it.  I don't think the whole implementation is very good.

ceres-c, 2018-02-14:
Theoretically the tag should stay in selected state until power cycled or wrong command is issued. If the pm3 cuts off power to antenna after a command then the tag will not be selected anymore, so this might be the reason. Do you know whether this is the case?

iceman1001, 2018-02-14:
this behavior is problematic to say the least.  Having the antenna running constantly if forgotten to turn off. 

the select command can be used for  **hf 15 raw** command but otherwise,   plus somehow notify user to turn off...      same goes with **hf 14a raw**...

ceres-c, 2018-02-14:
I personally don't have any issue about this, I haven't ever used this command and I don't need it.
Maybe there are some use cases which need selected mode, but to my knowledge it's interchangeable with addressed mode, so appending tag's UID it's enough to solve this problem.
To sum it up, hf 15 select could be removed...

iceman1001, 2018-02-14:
I guess I had an idea of making all  RAW commands act the same,  to seperate from **reader** command.
instead of guessing what this command do when trying another iso-protocol..

ceres-c, 2018-02-14:
Spring cleaning time is approaching, I take it ¯\\_(ツ)\_/¯

iceman1001, 2018-02-14:
go for it!

osysltd, 2018-02-15:
dears, for me I've got almost the same (tested on both RDV2 and Easy)

```
#db# [!] error, uneven octet! (extra bits!) mask 10          
crc fail          
...iso15693 card select failed          
iso15693 card select failed          
....#db# [!] error, uneven octet! (extra bits!) mask 20          
crc fail          
.#db# [!] error, uneven octet! (extra bits!) mask 20          
crc fail          
........iso15693 card select failed          
iso15693 card select failed          
.iso15693 card select failed          
#db# [!] error, uneven octet! (extra bits!) mask 40          
crc fail          
iso15693 card select failed          
iso15693 card select failed  
```

iceman1001, 2018-02-15:
latest iceman source?

osysltd, 2018-02-15:
yep, just build

```
[ ARM ]
 bootrom: /-suspect 2015-04-02 15:12:04
      os: iceman/master/ice_v3.1.0-585-g95c443c3 2018-02-15 10:00:31
 [ FPGA ]
 LF image built for 2s30vq100 on 2017/10/25 at 19:50:50
 HF image built for 2s30vq100 on 2017/11/10 at 19:24:16

```

iceman1001, 2018-02-15:
output from  **hf 15 list**  ?
I have tested on rdv2.0  and another pm3,  both work with tag on antenna and with distance

osysltd, 2018-02-15:
indeed, need a big distance (at least 3-5 cm)
works perfectly, thanks!

closed 2018-02-15 by iceman1001