iceman1001/proxmark3 #182: Proxmark3 rdv4.0

iceman1001, 2018-02-24:
Just wanted to do a shout-out for the new revision of proxmark3,   the pm3 rdv4.0

Go and back it up on kickstarter!
https://www.kickstarter.com/projects/1408815241/proxmark3-rdv-40

joanbono, 2018-02-24:
Good news!!!

Ordered!! 😄

iceman1001, 2018-02-24:
Yeah,  I saw your pledge!   You gonna love this one.  Its the same size as a Chameleon mini.  Just awesome!

tuxthemadpenguin, 2018-02-24:
anddd just as I thought I was happy with my toolkit.  :-D  Looks great iceman.

iceman1001, 2018-02-24:
@tuxthemadpenguin   sorry..     but this one is a nice one.

exploitagency, 2018-02-24:
What rewards include the "Smart card interface"?  And will it be sold separate and roughly what price point?

iceman1001, 2018-02-24:
The smart cards module is included on the pc. As a fundraiser we made it execlusive like the **pro release special** pledge.   If it is of interest we can add a pledge with it,   what do you think?

If / when we get funded,  I would expect that when the product hits the webshops, it most likely will be sold as a kit.  Where a basic kit would include converter,  case,  antenna.
And a pro kit would include the different sized antennas.
I have no idea what endprice the webshops will take,  but we aim in line with the pledges.
Those prices are actually great prices.

exploitagency, 2018-02-24:
Im feeling the pentester pro or the basic early bird with smart card interface since it is fairly unique.

iceman1001, 2018-02-24:
Its unique alright.      Still need to add support for it in the firmware.

exploitagency, 2018-02-24:
:)

exploitagency, 2018-02-24:
Congrats by the way, I shared the link to the kickstarter on my Twitter and hope I can figure out what package I want before they sell out, I am pretty confident the funding goals will be met.

iceman1001, 2018-02-24:
I hear ya and feel the love!  This one is a beauty!

exploitagency, 2018-02-25:
How can you pledge for multiple perks?

iceman1001, 2018-02-25:
I feel your love bro!

Yeah the Kevin Mitnick love echos very loud.

brantzau, 2018-02-25:
I knew you guys are working on something, but this is quick.

Congrats to make this real.

Pledged

joanbono, 2018-02-25:
Hi @iceman1001 ! Got friends asking me if the **Early Bird Package** comes with LF/HF antenna.

iceman1001, 2018-02-25:
Yes,  the antenna is an dual interface one.   

And here I just added a FAQ on kickstarter which answers some of the questions we have got so far.

[](https://www.kickstarter.com/projects/1408815241/proxmark3-rdv-40/faqs#project-faqs)

iceman1001, 2018-03-30:
....
when it comes to firmware,  I think I will just use iceman fork for it.
while the rdv40 will work on offical pm3 firmware, the new function and enhancements will not.
...

iceman1001, 2018-05-03:
Sweet,   uploading and downloading to the added 256kb flashmemory works.
All under the new command group **mem**

- ** mem load h **
- ** mem save h **
- ** mem wipe h **

iceman1001, 2018-05-06:
We could save some ARM flashmemory by moving the fpga images to the external flashmemory.
The compressed fpga_all.bit.z is 38kb in size.   Just an idea.

iceman1001, 2018-07-09:
Showing of the smart card (contact mode)  with sim card module and ISO7618 in this clip 
[pm3 rdv40 smartcard support](https://youtu.be/bYU7j3mwwDk)

iceman1001, 2018-07-09:
The new added support command is:
- **sc info**
- **sc reader**
- **sc list**
- **sc raw**
- **sc setclock**

You can also verify it, with  **hw status**

iceman1001, 2018-07-09:
@merlokk  you will find this interesting :)

closed 2018-07-10 by iceman1001