iceman1001/proxmark3 #185: hf_colin.c fails to compile

tuxthemadpenguin, 2018-02-25:
Current version of this repo, replace `-DWITH_HF_SAMYRUN \ ` with `-DWITH_HF_COLIN \` and `hf_young.c` with `hf_colin.c` in Makefile under armsrc/ .  Ran make clean && make all and I get :
```
arm-none-eabi-gcc -c -I../include -I../common -Wall -Werror -pedantic -Wunused -std=c99 -DWITH_CRC -DWITH_ISO18092 -DON_DEVICE -DWITH_LF -DWITH_HITAG -DWITH_ISO15693 -DWITH_LEGICRF -DWITH_ISO14443b -DWITH_ISO14443a -DWITH_ICLASS -DWITH_FELICA -DWITH_HFSNOOP -DWITH_HF_COLIN -fno-strict-aliasing -ffunction-sections -fdata-sections -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED -I../zlib -I. -Os -mthumb-interwork -o obj/hf_colin.o ../armsrc/Standalone/hf_colin.c 
../armsrc/Standalone/hf_colin.c: In function 'RunMod':
../armsrc/Standalone/hf_colin.c:201:33: error: pointer targets in passing argument 1 of 'sprintf' differ in signedness [-Werror=pointer-sign]
                                 foundKey[t][sectorNo][3], foundKey[t][sectorNo][4], foundKey[t][sectorNo][5]);
                                 ^
In file included from /usr/include/newlib/wchar.h:4:0,
                 from ../common/usb_cdc.h:38,
                 from ../common/cmd.h:38,
                 from ./iso14443a.h:21,
                 from ../armsrc/Standalone/hf_colin.h:15,
                 from ../armsrc/Standalone/hf_colin.c:11:
/usr/include/newlib/stdio.h:221:5: note: expected 'char * restrict' but argument is of type 'uint8_t *'
 int _EXFUN(sprintf, (char *__restrict, const char *__restrict, ...)
     ^
../armsrc/Standalone/hf_colin.c:208:29: error: pointer targets in passing argument 1 of 'sprintf' differ in signedness [-Werror=pointer-sign]
                             foundKey[t][sectorNo][3], foundKey[t][sectorNo][4], foundKey[t][sectorNo][5]);
                             ^
In file included from /usr/include/newlib/wchar.h:4:0,
                 from ../common/usb_cdc.h:38,
                 from ../common/cmd.h:38,
                 from ./iso14443a.h:21,
                 from ../armsrc/Standalone/hf_colin.h:15,
                 from ../armsrc/Standalone/hf_colin.c:11:
/usr/include/newlib/stdio.h:221:5: note: expected 'char * restrict' but argument is of type 'uint8_t *'
 int _EXFUN(sprintf, (char *__restrict, const char *__restrict, ...)
     ^
../armsrc/Standalone/hf_colin.c:233:29: error: pointer targets in passing argument 1 of 'sprintf' differ in signedness [-Werror=pointer-sign]
                             foundKey[t][sectorNo][3], foundKey[t][sectorNo][4], foundKey[t][sectorNo][5]);
                             ^
In file included from /usr/include/newlib/wchar.h:4:0,
                 from ../common/usb_cdc.h:38,
                 from ../common/cmd.h:38,
                 from ./iso14443a.h:21,
                 from ../armsrc/Standalone/hf_colin.h:15,
                 from ../armsrc/Standalone/hf_colin.c:11:
/usr/include/newlib/stdio.h:221:5: note: expected 'char * restrict' but argument is of type 'uint8_t *'
 int _EXFUN(sprintf, (char *__restrict, const char *__restrict, ...)
     ^
../armsrc/Standalone/hf_colin.c:238:29: error: pointer targets in passing argument 1 of 'sprintf' differ in signedness [-Werror=pointer-sign]
                             foundKey[t][sectorNo][3], foundKey[t][sectorNo][4], foundKey[t][sectorNo][5]);
                             ^
In file included from /usr/include/newlib/wchar.h:4:0,
                 from ../common/usb_cdc.h:38,
                 from ../common/cmd.h:38,
                 from ./iso14443a.h:21,
                 from ../armsrc/Standalone/hf_colin.h:15,
                 from ../armsrc/Standalone/hf_colin.c:11:
/usr/include/newlib/stdio.h:221:5: note: expected 'char * restrict' but argument is of type 'uint8_t *'
 int _EXFUN(sprintf, (char *__restrict, const char *__restrict, ...)
     ^
../armsrc/Standalone/hf_colin.c:243:29: error: pointer targets in passing argument 1 of 'sprintf' differ in signedness [-Werror=pointer-sign]
                             foundKey[t][sectorNo][3], foundKey[t][sectorNo][4], foundKey[t][sectorNo][5]);
                             ^
In file included from /usr/include/newlib/wchar.h:4:0,
                 from ../common/usb_cdc.h:38,
                 from ../common/cmd.h:38,
                 from ./iso14443a.h:21,
                 from ../armsrc/Standalone/hf_colin.h:15,
                 from ../armsrc/Standalone/hf_colin.c:11:
/usr/include/newlib/stdio.h:221:5: note: expected 'char * restrict' but argument is of type 'uint8_t *'
 int _EXFUN(sprintf, (char *__restrict, const char *__restrict, ...)
     ^
../armsrc/Standalone/hf_colin.c:248:29: error: pointer targets in passing argument 1 of 'sprintf' differ in signedness [-Werror=pointer-sign]
                             foundKey[t][sectorNo][3], foundKey[t][sectorNo][4], foundKey[t][sectorNo][5]);
                             ^
In file included from /usr/include/newlib/wchar.h:4:0,
                 from ../common/usb_cdc.h:38,
                 from ../common/cmd.h:38,
[... 120 lines cut ...]
                 from ../armsrc/Standalone/hf_colin.c:11:
/usr/include/newlib/stdio.h:221:5: note: expected 'char * restrict' but argument is of type 'uint8_t *'
 int _EXFUN(sprintf, (char *__restrict, const char *__restrict, ...)
     ^
../armsrc/Standalone/hf_colin.c: In function 'cjat91_saMifareChkKeys':
../armsrc/Standalone/hf_colin.c:554:14: error: unused variable 'cuid' [-Werror=unused-variable]
     uint32_t cuid;
              ^
../armsrc/Standalone/hf_colin.c:553:13: error: unused variable 'uid' [-Werror=unused-variable]
     uint8_t uid[10];
             ^
../armsrc/Standalone/hf_colin.c: In function 'saMifareCSetBlock':
../armsrc/Standalone/hf_colin.c:657:14: error: unused variable 'cuid' [-Werror=unused-variable]
     uint32_t cuid;
              ^
../armsrc/Standalone/hf_colin.c:655:13: error: unused variable 'uid' [-Werror=unused-variable]
     uint8_t uid[10] = {0x00};
             ^
cc1: all warnings being treated as errors
../common/Makefile.common:80: recipe for target 'obj/hf_colin.o' failed
make[1]: *** [obj/hf_colin.o] Error 1
make[1]: Leaving directory '/home/test/iceman2/proxmark3/armsrc'
Makefile:35: recipe for target 'armsrc/all' failed
make: *** [armsrc/all] Error 2
```

iceman1001, 2018-02-25:
did you also change inside  appmain.c   (near the end of the file) ?

iceman1001, 2018-02-25:
@tuxthemadpenguin  did you manage to compile?

tuxthemadpenguin, 2018-02-25:
Yes, it compiles.  Thank you!  I will test the functionality in a couple hours.

closed 2018-02-25 by tuxthemadpenguin