iceman1001/proxmark3 #186: [feature request] add wiegand decoding to proxmark3 client

thesle3p, 2018-02-26:
Hi would it be possible to add a wiegand decoder to the proxmark3 client (maybe under analyse) where you could specify bit length and offsets/lenthgs for FC and Card #  and it could return FC and card number from raw wiegand code? Perhaps  have things like cmdlfawid utilize that  to make the process of dealing with  awid credentials with unusual bit lengths easier for example (replace awid with any other card type that occasionally has a non standard bit length).

W8M2Hg9lLmWqXSGC, 2018-02-27:
This question has come up before. Being able to translate this requires a small database consisting of the various Physical Access Control System (PACS) formats.
Throwing a stream of random bits at a decoder would require user input as there will be a number of false positives.
Have you had a look at [Cardinfo](cardinfo.barkweb.com.au)? What do you think of the results produced?

thesle3p, 2018-02-27:
What's wrong with user input? This android app https://play.google.com/store/apps/details?id=com.muweili.cardformatcalculator&hl=en does what I am talking about very well I just was thinking it would be nice to do so from the proxmark client.

iceman1001, 2018-02-27:
I did have a wiegand decoder in **lf hid** before,  removed it since it didn't generate the right raw hex.

The easy part is to make a wiegand calc,.   The hard part is to generate the raw hax bytes needed for the clone commands.   All systems has their own quirks/ encoding layers above the wiegand layer.
 and at the laster layer,  is what I call the pm3 layer,  which adds the preamble and other stuff.

So people gets confused and thinks its just wiegand with the implemented dedicated systems like  awid, hid, nedap etc etc.    Behind the scens, it is not.

iceman1001, 2018-02-27:
@thesle3p   if you are good programmer, you can enable the previous wiegand calc in lf hid..  
and move it to [b]analyze wiegand[/b]    the wiegand part works

iceman1001, 2019-09-19:
I took the wiegand encoder/decoder from offical repo and added it to RRG/Iceman,  since this repo is deprecated.  

ref
https://github.com/RfidResearchGroup/proxmark3

You find the new command
```
pm3->wiegand
```

closed 2019-09-19 by iceman1001