iceman1001/proxmark3 #192: [idea] hf 14a anti collision fuzzing

iceman1001, 2018-02-28:
Reading this blog post by Christopher Wade (@iskuri)  I think that the mentioned fuzzing of a HF reader anticollision should be quite simple to implement.  No need to add it to the hf 14a sim,  but a own command lik  **hf 14a antifuzz**   with like one parameter  for uid length.  (4,7,10),    would do the trick.

I'm impressed with building the whole chain,  from hardware to improving crypt1 implementation for a 8bit MCU,   @iskuri is really good. 

[building half a proxmark for 10$ ](https://www.pentestpartners.com/security-blog/building-half-a-proxmark-for-10/)

Iskuri, 2018-02-28:
I am really happy you guys saw my post! Cheers for the feedback on it, I plan on releasing source and circuit designs very soon, though I have a few minor quirks to iron out before I do so. If you have any queries about the fuzzing side I am happy to help, in my experience it has been as simple as messing with the load modulation though.

iceman1001, 2018-02-28:
Actually,   your skills in optimizing the crypto1 is needed in the Chameleon mini revE rebooted project...
The current implementation is too slow.   https://github.com/iceman1001/ChameleonMini-rebooted/issues/29
The crypto patch for RevG is too big. Doesn't fit the limits.   Feeling up for it?

When it comes to the PM3 version,  it is like you say,  it should just load modulate during our simulation answers.  I guess sending out ONES will do?..   I am most interested to see your stuff!

Also the cleaning of the signal etc,   would be needed for the FeliCa implementation that suffers today,  could use your love.   :)

Iskuri, 2018-02-28:
I'd definitely like to help if I can, unfortunately I embarrassingly don't own a Chameleon and have not looked at the code too much, but will take a look now, do you have any test harnesses I can use? 

As for the PM3, what I did was to just send the raw PWM signal constantly, so no manchester encoding or anything, and that seems to effectively work. I'm not too sure about the FeliCa stuff, I will go and look at it.

iceman1001, 2018-02-28:
Seeing what you can do,  I am very certain you will overachive success!  :)
Here I thought you had 'em all since you wrote about them in your blog.  haha. 
Send me an email,  (iceman  @ iuse se )  and we take it from there.

PM3,  yeah,  the command itself shouldn't be hard at all to implement.  Even simpler if you just load modulate during at anti-coll phase...  

FeliCa, again,  you are humble,  I like that.

Iskuri, 2018-02-28:
Thanks! I will definitely give it a go. Unfortunately I don't have any of them, though I believe I have friends who have Proxmarks, I am not sure. When I developed my tag I used the most cheap equipment I could find. I even programmed the ATTiny using an Arduino in ISP mode (though I used avr-gcc and avrdude for programming it), any complex debugging I needed to do on the crypto I did using a test harness in simavr so that might be helpful. Either way, I will shoot you an email soon. :)

iceman1001, 2018-02-28:
well, only if you are up for it,   if so,  I think I can fix a mini for you :)

Iskuri, 2018-02-28:
I'm definitely interested! I will shoot you an email to discuss.

iceman1001, 2018-02-28:
@Iskuri 
I made it easy for you,    
https://github.com/iceman1001/proxmark3/blob/master/armsrc/iso14443a.c#L1797
just add your stuff here ;) 

The command is hooked with  **hf 14a antifuzz**

Iskuri, 2018-02-28:
Awesome, I will definitely look at that, though I may need help testing it's functionality.