iceman1001/proxmark3 #195: Feature suggestion data directory parameter

tuxthemadpenguin, 2018-03-01:
It might be nice to have a config file Proxmark client can check for in the directory or a command line parameter to specify a default data directory for dumps, keyfiles, etc.  

proxmark3 --data=~/rfData
or 
a variable in a config.

1) It can help decrease the clutter a bit in the client directory.

2) I am switching between a couple forks,  it would be nice to keep all my different key files and dumps together.

I know I can do this manually by moving data around afterwards, using relative paths, or absolute paths.  It just might make it a bit nicer, especially for the docker image.

iceman1001, 2018-03-01:
That is an execellent idea.

iceman1001, 2019-09-16:
Hi Woody,
If you switch to RRG/Iceman repo,  you will find this is implemented. 
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/path_notes.md

closed 2019-09-16 by iceman1001