iceman1001/proxmark3 #205: hf 14b SRI tag reading currently broken

elafargue, 2018-03-15:
On git hash a7fd398d332e9b76 

SRI512 tag detection and reading is currently not working at all.

Did A/B test between master repo and your fork, and all SRI-related functions seem to be not working on your fork a the moment, so I though I'd tag this so that you can take a look in case you didn't know about it already?

elafargue, 2018-03-15:
I also noticed that generic detection of 14b tags seems to be broken as well, which makes since since SRI tags use a 14b modulation. Tried with another 14b card (Swedish id card prototype) and it can be detected fine on main repo, but not at all on this fork.

iceman1001, 2018-03-15:
yup, yup,   14b suffers at the moment.

iceman1001, 2018-06-21:
hf 14b reader works again.

closed 2018-06-21 by iceman1001