iceman1001/proxmark3 #210: Flashing bricked pm3 elechouse, failing to recover from segger

samyk, 2018-04-06:
Hola! I tried updating the latest bootrom, compiled from git today, which first worked, but the fullimage flash failed. I tried updating both again and that failed and pm3 was not responsive anymore, though the serial port was still visible.

I then connected a Segger Jlink and was able to communicate with the MCU and flash recovery/proxmark3_recovery.bin, but the pm3 does not show up as a serial device anymore. I also tried flashing recovery/bootrom.bin and recovery/fullimage.bin, but still cannot see it as a serial device. Thoughts?

Thanks!

```
SEGGER J-Link Commander V6.14a (Compiled Feb 27 2017 18:40:38)
DLL version V6.14a, compiled Feb 27 2017 18:40:25

Connecting to J-Link via USB...O.K.
Firmware: J-Link V9 compiled Jun 16 2017 16:15:10
Hardware version: V9.40
S/N: 269402326
License(s): FlashBP, GDB
OEM: SEGGER-EDU
VTref = 3.172V

Type "connect" to establish a target connection, '?' for help
J-Link>connect
Please specify device / core. <Default>: AT91SAM7S256
Type '?' for selection dialog
Device>AT91SAM7S512
Please specify target interface:
  J) JTAG (Default)
TIF>J
Device position in JTAG chain (IRPre,DRPre) <Default>: -1,-1 => Auto-detect
JTAGConf>
Specify target interface speed [kHz]. <Default>: 4000 kHz
Speed>200
Device "AT91SAM7S512" selected.

TotalIRLen = 4, IRPrint = 0x01
Found 1 JTAG device, Total IRLen = 4:
 #0 Id: 0x3F0F0F0F, IRLen: 04, IRPrint: 0x1, ARM7TDMI Core
ARM7 identified.
J-Link>h
PC: (R15) = 00112FB8, CPSR = 000000F3 (SVC mode, THUMB FIQ dis. IRQ dis.)
Current:
     R0 =000003E8, R1 =00112FBD, R2 =00000300, R3 =FFFFFD40
     R4 =0020FFE0, R5 =00000002, R6 =80000000, R7 =40000020
     R8 =80000800, R9 =09004000, R10=01000200, R11=44129000, R12=0003D876
     R13=0020FD58, R14=00112FA7, SPSR=E00000FE
USR: R8 =80000800, R9 =09004000, R10=01000200, R11=44129000, R12=0003D876
     R13=10000000, R14=00000010
FIQ: R8 =00000010, R9 =00800420, R10=02008920, R11=84000004, R12=80008080
     R13=00100000, R14=80081000, SPSR=B00000FF
IRQ: R13=00010008, R14=00080001, SPSR=F00000FA
SVC: R13=0020FD58, R14=00112FA7, SPSR=E00000FE
ABT: R13=00000010, R14=80480120, SPSR=F00000FF
UND: R13=10000009, R14=00000800, SPSR=B000007F
J-Link>loadbin /Users/samy/Code/proxmark/pm3-iceman/recovery/proxmark3_recovery.bin 0x100000
Downloading file [/Users/samy/Code/proxmark/pm3-iceman/recovery/proxmark3_recovery.bin]...
Comparing flash   [100%] Done.
Verifying flash   [100%] Done.
J-Link: Flash download: Flash download skipped. Flash contents already match
O.K.
J-Link>loadbin /Users/samy/Code/proxmark/pm3-iceman/recovery/bootrom.bin 0x100000
Downloading file [/Users/samy/Code/proxmark/pm3-iceman/recovery/bootrom.bin]...
Comparing flash   [100%] Done.
Verifying flash   [100%] Done.
J-Link: Flash download: Flash download skipped. Flash contents already match
O.K.
J-Link>loadbin /Users/samy/Code/proxmark/pm3-iceman/recovery/fullimage.bin 0x102000
Downloading file [/Users/samy/Code/proxmark/pm3-iceman/recovery/fullimage.bin]...
Comparing flash   [100%] Done.
Verifying flash   [100%] Done.
J-Link: Flash download: Flash download skipped. Flash contents already match
O.K.
J-Link>
```

iceman1001, 2018-04-06:
Strange but your segger says it skipped flashing since contents already match..
...
Which OS are you running on?

samyk, 2018-04-06:
That was actually a reattempt at flashing. The first time I flashed it via
segger it did in fact flash (not just verify), so I ran it again to confirm
it was written correctly and to produce output to paste here.

Running macOS 10.12

samyk, 2018-04-06:
Here's the weird thing, I have an original Proxmark3 where the serial device **does** show up on the same machine. Additionally, the elechouse PM3 (bricked) does show up as a USB device. Tested on same USB port and tested multiple USB cables.

Bricked pm3 `system_profiler SPUSBDataType` output:
```
        PM3 Device:

          Product ID: 0x4b8f
          Vendor ID: 0x9ac4
          Version: 0.01
          Serial Number: 88888888
          Speed: Up to 12 Mb/sec
          Manufacturer: proxmark.org
          Location ID: 0x14100000 / 5
          Current Available (mA): 500
          Current Required (mA): 500
          Extra Operating Current (mA): 0
```

Original pm3 `system_profiler SPUSBDataType` output: 
```
        Communication Device:

          Product ID: 0x504d
          Vendor ID: 0x2d2d
          Version: 0.01
          Serial Number: proxmark.org
          Speed: Up to 12 Mb/sec
          Manufacturer: proxmark.org
          Location ID: 0x14100000 / 31
          Current Available (mA): 500
          Current Required (mA): 500
          Extra Operating Current (mA): 0
```

samyk, 2018-04-07:
Will close this as it doesn't seem to pertain to this fork. If I load a Linux VM, PM3 properly shows up as a serial device and is usable there. The strange thing is it was working fine until I updated the bootrom, and flashing via jtag made it stop responding over serial on macOS, but downgrading the bootrom didn't bring it back either, so not quite sure what happened. At least it's usable on Linux!

closed 2018-04-07 by samyk

iceman1001, 2018-04-07:
Is this the latest source from github?  ...it looks like homebrew older version.
since the latest source only has  888  as serial number...

samyk, 2018-04-09:
The paste may have been from the version on the proxmark3 repo. I did also use the latest from this git repo and tried the original proxmark3 repo afterwards. I do in fact see "Serial 888" on a Linux VM.

samyk, 2018-04-10:
I tried manually unloading the extension `sudo kextunload /System/Library/Extensions/Proxmark3.kext`, removing the extension `sudo rm -rf /System/Library/Extensions/Proxmark3.kext`, and reinstalling the extension `sudo make install_kext`, including rebooting, to no avail.

I suspect another kext is taking over the device still.