iceman1001/proxmark3 #212: No response from Proxmark during hardnested attack

Caydenz, 2018-04-10:
`pm3 --> hf mf hard 0 A xxxxxxxxxxxx 4 A
--target block no:  4, target key type:A, known target key: 0x000000000000 (not set), file action: none, Slow: No, Tests: 0

 time    | #nonces | Activity                                                | expected to brute force
         |         |                                                         | #states         | time
------------------------------------------------------------------------------------------------------
       0 |       0 | Start using 4 threads and AVX2 SIMD core                |                 |
       0 |       0 | Brute force benchmark: 1116 million (2^30.1) keys/s     | 140737488355328 |   35h
       1 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |   35h
       4 |     112 | Apply bit flip properties                               |   1551001976832 | 23min
       6 |     224 | Apply bit flip properties                               |    622463352832 |  9min
       9 |     336 | Apply bit flip properties                               |    522774413312 |  8min
      12 |     441 | Apply bit flip properties                               |    503123902464 |  8min
[!] Error: No response from Proxmark.

pm3 -->`

When I'm trying to bruteforce a card, I get the following errors on my Proxmark3 Easy V3. Any idea what's happening?

iceman1001, 2018-04-10:
looks like your client fails...  Do you have enough ram memory in your computer?

Caydenz, 2018-04-10:
Yes...

```
 andrei@genisys  ~  neofetch
██████████████████  ████████   andrei@genisys
██████████████████  ████████   --------------
██████████████████  ████████   OS: Manjaro Linux x86_64
██████████████████  ████████   Host: 2325BA3 ThinkPad X230
████████            ████████   Kernel: 4.14.27-1-MANJARO
████████  ████████  ████████   Uptime: 1 hour, 9 mins
████████  ████████  ████████   Packages: 917
████████  ████████  ████████   Shell: zsh 5.4.2
████████  ████████  ████████   DE: GNOME
████████  ████████  ████████   Theme: Arc-Maia-Dark [GTK2/3]
████████  ████████  ████████   Icons: Papirus-Adapta-Maia [GTK2/3]
████████  ████████  ████████   Terminal: terminator
████████  ████████  ████████   CPU: Intel i5-3320M (4) @ 3.300GHz
████████  ████████  ████████   GPU: Intel Ivybridge Mobile
                               Memory: 1058MiB / 7680MiB

```
`file client/proxmark3
client/proxmark3: ELF 64-bit LSB shared object, x86-64, version 1 (GNU/Linux), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 3.2.0, BuildID[sha1]=5384b74ebf8d6fc3ad47697e47c037abf7219b70, with debug_info, not stripped
`

Tried on a Windows system with 16GB as well.

Caydenz, 2018-04-10:
So... I added some quick printfs in the code and it fails here (function acquire_nonces):

```
if (!initialize) {
	if (!WaitForResponseTimeout(CMD_ACK, &resp, 3000)) {
		if (nonce_file_write) {
			fclose(fnonces);
		}
		return 1;
	}
```
Should I try to increase the timeout?

iceman1001, 2018-04-10:
ok,  so your nonce collecting is hurting. 
Try the  -s -ss parameters to enable the "slow" aquisition  and see if that helps.

Don't forget to have some distance between antenna and tag.

Caydenz, 2018-04-10:
I tried with slow acquisition and some distance between tag and antenna :(
I'll try downgrading to "minor Sweet Lemon" release and see how that goes.

iceman1001, 2018-04-10:
hm..   are your udev/rules ok?

Caydenz, 2018-04-10:
I just ran "make udev" before compiling the repository. That's everything I have to do, right?

iceman1001, 2018-04-10:
that should do the trick. 
Could also be a bad device...  where did you buy it from?

Caydenz, 2018-04-10:
So.... I think I forgot to reboot after the "make udev" =) 
Anyway, I rebooted, compiled the "minor Sweet Lemon" release and it is now working!

Thank you so much for all the help! :)

iceman1001, 2018-04-10:
"flashed the master"?
Do you mean you went back and flashed latest iceman?  or did you mean you took latest offical repo?

Caydenz, 2018-04-10:
I flashed 2 releases:
1. The latest iceman release (git clone...) <--- hardnested is not working
2. minor Sweet Lemon release (v3.1.0) <--- hardnested is working

I got minor Sweet Lemon from here: https://github.com/iceman1001/proxmark3/releases/tag/ice_v3.1.0

iceman1001, 2018-04-10:
ok, would you mind flashing latest source from offical pm3 and test?

Caydenz, 2018-04-10:
Sure. I'll let you know when it's done. (probably tomorrow, but I'll try to get it done sooner)

Caydenz, 2018-04-10:
I tested the official pm3 and it works.
You can find below details of each image:

Official:
```
qt5ct: using qt5ct plugin
Prox/RFID mark3 RFID instrument    
bootrom: iceman// 2018-04-10 21:28:25
os: master/v3.0.1-361-ge069547-suspect 2018-04-10 19:50:54
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/10/27 at 08:30:59
    
uC: AT91SAM7S256 Rev D    
Embedded Processor: ARM7TDMI    
Nonvolatile Program Memory Size: 256K bytes. Used: 196805 bytes (75%). Free: 65339 bytes (25%).    
Second Nonvolatile Program Memory Size: None    
Internal SRAM Size: 64K bytes    
Architecture Identifier: AT91SAM7Sxx Series    
Nonvolatile Program Memory Type: Embedded Flash Memory  
```
Iceman fork:
```
qt5ct: using qt5ct plugin
Proxmark3 RFID instrument
               
 [ ARM ]
 bootrom: iceman// 2018-04-10 21:28:25
      os: iceman/master/ice_v3.1.0-787-g192aa9ab-dirty-unclean 2018-04-10 22:56:59
 [ FPGA ]
 LF image built for 2s30vq100 on 2017/10/25 at 19:50:50
 HF image built for 2s30vq100 on 2017/11/10 at 19:24:16
              
 [ Hardware ]               
  --= uC: AT91SAM7S256 Rev D    
  --= Embedded Processor: ARM7TDMI            
  --= Nonvolatile Program Memory Size: 256K bytes, Used: 234412 bytes (89%) Free: 27732 bytes (11%)    
  --= Second Nonvolatile Program Memory Size: None    
  --= Internal SRAM Size: 64K bytes    
  --= Architecture Identifier: AT91SAM7Sxx Series    
  --= Nonvolatile Program Memory Type: Embedded Flash Memory    

```

Caydenz, 2018-04-15:
I tested a bunch of commits and it seems the latest commit is causing this problem. (The SmartCard commit)

iceman1001, 2018-04-16:
nice,  that part of the code is under construction,  but I pushed some changes meanwhile in order for the hardnest to work.  It works during my tests at least.

Caydenz, 2018-04-16:
Just tested your latest commits and hardnested is working. Any tips where I should get some knowledge about all this stuff? (embedded stuff)

My knowledge of embedded C is somewhat limited. (simple C is pretty good :D)

Thanks!

iceman1001, 2018-04-16:
to read up on embedded stuff?  I suggest you use your google-foo and start from there.
Download the datasheet for the embedded device you want use and see if there is any sample code ...

closed 2018-04-16 by iceman1001