iceman1001/proxmark3 #213: Segmentation fault - macOS High Sierra

jjmuriel, 2018-04-11:
Hi,

I have this problem in my Macbook with macOS  version 10.13.4 (17E199):

pm3 --> hf mf hard 0 A 7b296f353c6b 0 b
--target block no:  0, target key type:B, known target key: 0x000000000000 (not set), file action: none, Slow: No, Tests: 0           

          
 time    | #nonces | Activity                                                | expected to brute force          
         |         |                                                         | #states         | time           
------------------------------------------------------------------------------------------------------          
       0 |       0 | Start using 8 threads and AVX2 SIMD core                |                 |          
       0 |       0 | Brute force benchmark: 629 million (2^29.2) keys/s      | 140737488355328 |    3d          
       2 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |    3d          
      17 |     112 | Apply bit flip properties                               |     31871776768 |   51s          
      28 |     224 | Apply bit flip properties                               |      2864715008 |    5s          
      39 |     335 | Apply bit flip properties                               |      1405592064 |    2s          
      50 |     447 | Apply bit flip properties                               |      1405592064 |    2s          
      70 |     557 | Apply bit flip properties                               |      1405592064 |    2s          
      81 |     668 | Apply bit flip properties                               |      1405592064 |    2s          
     101 |     777 | Apply bit flip properties                               |      1405592064 |    2s          
     111 |     888 | Apply bit flip properties                               |      1405592064 |    2s          
     132 |    1000 | Apply bit flip properties                               |      1405592064 |    2s          
     152 |    1108 | Apply bit flip properties                               |      1405592064 |    2s          
     172 |    1217 | Apply bit flip properties                               |      1405592064 |    2s          
     192 |    1327 | Apply bit flip properties                               |      1405592064 |    2s          
     214 |    1437 | Apply Sum property. Sum(a0) = 120                       |        26452860 |    0s          
     234 |    1437 | (1. guess: Sum(a8) = 192)                               |        26452860 |    0s          
Segmentation fault: 11

This is the the trace from my Macbook:

Process:               proxmark3 [25550]
Path:                  /usr/local/bin/proxmark3
Identifier:            proxmark3
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        bash [450]
Responsible:           proxmark3 [25550]
User ID:               501

Date/Time:             2018-04-11 10:37:43.522 -0500
OS Version:            Mac OS X 10.13.4 (17E199)
Report Version:        12
Anonymous UUID:        8DFB7D5B-E4AF-5F75-5420-C824AD23C3B5

Time Awake Since Boot: 4300 seconds

System Integrity Protection: enabled

Crashed Thread:        6

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x000000010f0d5000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [0]

VM Regions Near 0x10f0d5000:
    MALLOC_LARGE           000000010eed5000-000000010f0d5000 [ 2048K] rw-/rwx SM=PRV  
--> 
    MALLOC_LARGE           000000010fad5000-000000010fed5000 [ 4096K] rw-/rwx SM=PRV  

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff5dbc020a mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff5dbbf724 mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff357647d5 __CFRunLoopServiceMachPort + 341
3   com.apple.CoreFoundation      	0x00007fff35763b27 __CFRunLoopRun + 1783
[... 360 lines cut ...]
__TEXT                           150.7M      234 
__UNICODE                          560K        2 
mapped file                       50.8M       16 
shared memory                      744K       13 
===========                     =======  ======= 
TOTAL                              1.8G      583 

Model: MacBookPro11,2, BootROM MBP112.0145.B00, 4 processors, Intel Core i7, 2.2 GHz, 16 GB, SMC 2.18f15
Graphics: Intel Iris Pro, Intel Iris Pro, Built-In
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x134), Broadcom BCM43xx 1.0 (7.77.37.29.1a7)
Bluetooth: Version 6.0.5f3, 3 services, 27 devices, 1 incoming serial ports
Network Service: Thunderbolt Ethernet, Ethernet, en5
PCI Card: Apple 57762-A0, Ethernet Controller, Thunderbolt@195,0,0
Serial ATA Device: APPLE SSD SM0256F, 251 GB
USB Device: USB 3.0 Bus
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: PM3 Device
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.1
Thunderbolt Device: Thunderbolt to Gigabit Ethernet Adapter, Apple Inc., 1, 5.5

Can you help me , please?

jjmuriel, 2018-04-11:
This is the information about my Proxmark3:

MacBook-Pro:3.1.0 jjmuriel$ proxmark3 /dev/cu.usbmodem881 

Proxmark3 RFID instrument
          
 [ ARM ]
 bootrom: iceman/master/ice_v3.1.0-787-g192aa9a-dirty-unclean 2018-04-11 00:36:05
      os: iceman/master/ice_v3.1.0-787-g192aa9a-dirty-unclean 2018-04-11 00:36:08
 [ FPGA ]
 LF image built for 2s30vq100 on 2017/10/25 at 19:50:50
 HF image built for 2s30vq100 on 2017/11/10 at 19:24:16
          
 [ Hardware ]           
  --= uC: AT91SAM7S256 Rev D          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 256K bytes, Used: 234523 bytes (89) Free: 27621 bytes (11)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

jjmuriel, 2018-04-11:
If I read the nonces, this is that happen:

pm3 --> hf mf hard r 7b296f353c6b
--target block no:  0, target key type:A, known target key: 0x7b296f353c6b, file action: read, Slow: No, Tests: 0           

          
 time    | #nonces | Activity                                                | expected to brute force          
         |         |                                                         | #states         | time           
------------------------------------------------------------------------------------------------------          
       0 |       0 | Start using 8 threads and AVX2 SIMD core                |                 |          
       0 |       0 | Brute force benchmark: 626 million (2^29.2) keys/s      | 140737488355328 |    3d          
       2 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |    3d          
       6 |       0 | Reading nonces from file nonces.bin...                  | 140737488355328 |    3d          
       6 |    1792 | Read 1792 nonces from file. cuid=5b3335b7               | 140737488355328 |    3d          

BUG: known target key's even state is not member of first nonce byte's (0x88) states_bitarray!

BUG: known target key's odd  state is not member of first nonce byte's (0x88) states_bitarray!

BUG: known target key's even state is not member of all_bitflips_bitarray!

BUG: known target key's odd  state is not member of all_bitflips_bitarray!
      13 |    1792 | (1. guess: Sum(a8) = 0)                                 |       819276672 |    1s          
Segmentation fault: 11

jjmuriel, 2018-04-11:
[image]

iceman1001, 2018-04-16:
Sorry,  I can't help on OSX. My suggestion is that you search older issues and on the proxmark3 forum for help.

closed 2018-04-16 by jjmuriel