iceman1001/proxmark3 #218: proxmark3 rdv4 boot/load.

MJGrey, 2018-07-24:
I'm not sure whether this is a problem but judging from the video you posted recently, you didn't have this output. I seem to get it though and I'm not well versed enough yet to judge whether its an issue or not:

Proxmark3 RFID instrument
          
 [ ARM ]
 bootrom: iceman/master/ice_v3.1.0-873-gc15e755e-dirty-unclean 2018-05-26 08:47:16
      os: iceman/master/ice_v3.1.0-873-gc15e755e-dirty-unclean 2018-05-26 08:47:27
 [ FPGA ]
 LF image built for 2s30vq100 on 2017/10/25 at 19:50:50
 HF image built for 2s30vq100 on 2017/11/10 at 19:24:16
          
 [ Hardware ]           
  --= uC: AT91SAM7S512 Rev B          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 238314 bytes (45%) Free: 285974 bytes (55%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory          

          
#db# unknown command:: 0x0141          
#db# unknown command:: 0x0154          
#db# unknown command:: 0x0145          
#db# unknown command:: 0x0131          
#db# FlashMem read | 0 - 0          
#db# unknown command:: 0x0145          
#db# unknown command:: 0x0130          
#db# unknown command:: 0x0141          
#db# unknown command:: 0x0154          
#db# unknown command:: 0x0145          
#db# unknown command:: 0x0131          
#db# FlashMem read | 0 - 0          
#db# unknown command:: 0x0145          
#db# unknown command:: 0x0130          
#db# unknown command:: 0x010d          
#db# unknown command:: 0x0141          
#db# unknown command:: 0x0154          
#db# unknown command:: 0x0145          
#db# unknown command:: 0x0131          
#db# FlashMem read | 0 - 0          
#db# unknown command:: 0x0145          
#db# unknown command:: 0x0130          
#db# unknown command:: 0x010d          
#db# unknown command:: 0xf078007e          
pm3 -->

iceman1001, 2018-07-24:
and that looks like someone is not using the same client / firmware from same compilation

MJGrey, 2018-07-24:
So I'm going to have to update/flash it?

MJGrey, 2018-07-25:
Started doing that about 30 min ago (I didn't pay attention at first but now I saw the update.sh). It seems to be stuck on a part or its just taking its time.

iceman1001, 2018-07-25:
it shouldn't take very long time to flash.  break it with ctrl-c,    lookup your comport enumeration,  unplug your device,    press button and keep it pressed,   plug in the cable,   start the flasher again...

iceman1001, 2018-07-25:
pull latest code,  I updated the update.sh  script.

MJGrey, 2018-07-25:
Ok , I got this: 

Waiting for Proxmark to appear...
Found proxmark on /dev/pm3-0

./update.sh: line 14: client/flasher: No such file or directory

so I checked the update.sh script to see which directory it was linking to and indeed, atm all the files in the armsrc/obj/ folder are gone, replaced with Standalone/ which is also empty.

MJGrey, 2018-07-25:
ps sorry to be a bother, I'm sure you've got better things to be doing than noob-sitting haha, its much appreciated though.

iceman1001, 2018-07-25:
You need to compile the source.

MJGrey, 2018-07-25:
Yeah I just realized i had to do all that again, already deleted my comment, sorry about that.

iceman1001, 2018-07-25:
So we can now leave this subject closed ;)    We are all beginners at sometime.  Good thing to remember.