iceman1001/proxmark3 #224: Hardnested crashing the client on 2nd run.

TomHarkness, 2018-08-10:
Hi guys,

Latest iceman FW and client on an RDV2 - hardnested crashes the client on the second attempt.

Strangely this is not happening with a proxmark3 easy on the same build.

Latest OS X with an 8th gen i7 with 32GB of memory so it is not resource relates like some previous similar issues.

iceman1001, 2018-08-10:
...opened and closed.  I guess it does work ?

TomHarkness, 2018-08-13:
Issue was with a specific card so closed until I know more. Throws errors
every time when trying to hardnest block 11 key B. Every single time with
that same card. Other cards work fine.

iceman1001, 2018-08-13:
Can you save a nonce file from when running hardnested against that sector?

TomHarkness, 2018-08-13:
Will do this and get it uploaded for you later today.

TomHarkness, 2018-08-15:
Update: I think this is actually happening when a card is too close to the HF ant on my RDV3

iceman1001, 2018-08-15:
1-2cm distance is usually needed for 14a comms.   All 14a comms works very bad if tag is direct on antenna.