iceman1001/proxmark3 #225: hf mf chk key type reversed

TomHarkness, 2018-08-10:
Latest Client and firmware see below output for details. Checks key A when B is input and B when A is input:

pm3 --> hf mf chk 0 A 2A2C13CC242A
[ 0] key 2A 2C 13 CC 24 2A
.
Time in checkkeys: 0 seconds

testing to read key B...
|---|----------------|---|----------------|---|
|sec|key A           |res|key B           |res|
|---|----------------|---|----------------|---|
|000|  ------------  | 0 |  ------------  | 0 |
|---|----------------|---|----------------|---|

pm3 --> hf mf chk 0 B 2A2C13CC242A
[ 0] key 2A 2C 13 CC 24 2A
.
Time in checkkeys: 0 seconds

|---|----------------|---|----------------|---|
|sec|key A           |res|key B           |res|
|---|----------------|---|----------------|---|
|000|  2a2c13cc242a  | 1 |  ------------  | 0 |
|---|----------------|---|----------------|---|

iceman1001, 2018-08-10:
Confirmed,  good catch!

iceman1001, 2018-08-10:
Try presented fix..

closed 2018-08-10 by iceman1001

TomHarkness, 2018-08-13:
Thanks you beat me too it! Been flat out with other work. Legend!

commit 9b03cd9c (2018-08-10): Update cmdhfmf.c

fix #225