iceman1001/proxmark3 #228: Flashed Firmware - Now I have 4 solid red lights and wont recognize it

original13, 2018-08-13:
I ran following. It hung at  0x00100200..0x00100f0b [0xd0c / 7 blocks]....     
lights turned red and now it wont recognize it. I was holding button down whole time.
Where do I go from here?

sudo proxmark3-flasher /dev/cu.usbmodem881 -b /usr/local/Cellar/proxmark3/3.1.0/share/firmware/bootrom.elf /usr/local/Cellar/proxmark3/3.1.0/share/firmware/fullimage.elf 
Loading ELF file '/usr/local/Cellar/proxmark3/3.1.0/share/firmware/bootrom.elf'...
Loading usable ELF segments:
0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
1: V 0x00200000 P 0x00100200 (0x00000d0c->0x00000d0c) [R X] @0x298

Loading ELF file '/usr/local/Cellar/proxmark3/3.1.0/share/firmware/fullimage.elf'...
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x00031650->0x00031650) [R X] @0x94
1: V 0x00200000 P 0x00133650 (0x00001a14->0x00001a14) [RW ] @0x316e4
Note: Extending previous segment from 0x31650 to 0x33064 bytes

Waiting for Proxmark to appear on /dev/cu.usbmodem881. Found.

Flashing...
Writing segments for file: /usr/local/Cellar/proxmark3/3.1.0/share/firmware/bootrom.elf
 0x00100000..0x001001ff [0x200 / 1 blocks]. OK
 0x00100200..0x00100f0b [0xd0c / 7 blocks]....

iceman1001, 2018-08-13:
Don't know much of OSX but the wiki section for MacOS has some info about kext etc.
https://github.com/Proxmark/proxmark3/wiki/MacOS

@TomHarkness also mentioned something about very cheap USB-C --> USB adaptor could spook things.

And MacOS ppl seem to have a hard time in general...

https://github.com/Proxmark/proxmark3/issues/650
https://github.com/Proxmark/proxmark3/issues/633
https://github.com/Proxmark/proxmark3/issues/627
https://github.com/Proxmark/proxmark3/issues/283

original13, 2018-08-13:
@iceman1001... Thanks. I'll give it a try!

TomHarkness, 2018-08-13:
Will try this myself with the brew tap and report back.

For future reference there is no need to hold the button while flashing unless something goes wrong in the first place or, you are in the process of moving from HID to CDC or vice versa which, if you are on the RDV4 or recent RDV3 or easy hardware there should be no need.

iceman1001, 2018-08-17:
@original13    Using the following scripts on MacOS has helped other ppl.

[code]
./install.sh            -- installs a dev-env.
./update.sh          -- flashes your device with the first enumerated pm3 device found
./proxmark.sh       -- runs the proxmark3 client with the first enumerated pm3 device found
[/code]

TomHarkness, 2018-08-30:
How is the status of this? Any news?

original13, 2018-09-05:
I ordered and finally received a JTAG tool and Segger link. I reflashed the firmware and bootloader using that and now it's working fine.

Thanks all...!

closed 2018-09-05 by original13