iceman1001/proxmark3 #235: Can't run amiibo lua script

axfla, 2018-08-22:
Hi
Running amiibo lua script resulted in "libcrypto.so.1.0.0 cannot open shared object file no such file or directory", so I had to downgrade to libssl1.0.0 in order to get this old shared library.
I am now facing "./libluamiibo.so: undefined symbol: lua_createtable" when trying to run it and have no clue on how to fix it. Is the amiibo script deprecated, or should it still work ?
Thanks

iceman1001, 2018-08-22:
nay,  its just merged from james repo,  however his repo has another setting in his client makefile which allows for dynamic loading of libs. 
you find it here:
https://github.com/nccgroup/proxmark3-amiimicyou/blob/master/client/Makefile#L162

I didn't get it to compile on mingw, so feel free to fix it.

closed 2018-08-22 by axfla

iceman1001, 2018-08-22:
he also has a different view on bigbuff memory,  so it will need adaptations on it.
To eload it up,  don't need all reads like he did it.