iceman1001/proxmark3 #239: Fixed problem samyrun, but is it a good fix?

DidierStevens, 2018-09-01:
I could not get samyrun on my new RDV4 to read HID cards.

I Debugged it: the card is never read because signalprop.isnoise is always TRUE.

My fix: in common/lfdemod.c, line 1892, replace
    if (signalprop.isnoise) return -2;
with
    if (justNoise(dest, *size)) return -2;

Now I can read HID cards in standalone mode.

But is this a good fix? Because I compared with the Proxmark repository, and this function (HIDdemodFSK) has changed significantly.
Is it better to copy that new function?

TomHarkness, 2018-09-02:
Nice Find!

iceman1001, 2018-09-02:
Cool!  I believe the guys behind project walrus mentioned this fix some weeks ago.   Thought that they were going to push PR.   This fixes the current problem.

iceman1001, 2018-09-05:
I pushed fixes into RRG repo.

osysltd, 2018-09-05:
@iceman1001 can you please share the link to RRG repo?

iceman1001, 2018-09-05:
Of course.
https://github.com/RfidResearchGroup/proxmark3

iceman1001, 2018-09-13:
@DidierStevens   does the standalone work for you now?  time to close?

DidierStevens, 2018-09-13:
@iceman1001  I'll check tomorrow.

iceman1001, 2018-09-16:
@DidierStevens  its quite easy to swap which standalone that compiles in that repo now.
Only one line change in armsrc/makefile.    Don't let that stop you.

DidierStevens, 2018-09-20:
The standalone in that repo works for me.

The line I had to change was in common/Makefile_Enabled_Options.common:
APP_CFLAGS += -DWITH_ISO14443a_StandAlone \