iceman1001/proxmark3 #240: An GUI problem

zywshuai, 2018-09-02:
I wrote the code like this,but i cant get the number behind the "block data:" please help

            -     Case "button_break"
-                     item.Control = New Button
-                     item.Control.Text = item.Text
-                     ToolTip1.SetToolTip(item.Control, item.Tooltip)
-                     AddHandler item.Control.Click, AddressOf hi
-                     AddHandler item.Control.Click, AddressOf actionCommand
-                     'AddHandler hi(), AddressOf actionCommand
-                     'sector = hi()
-                     Call hi()
-                     item.Action &= ("hf mf nested 1" & sector)
-                     item.Control.Tag = item.Action
-                     item.Control.SetBounds(0, i * 35, 140, 20)
-                     item.Control.AutoSize = True

-     Public Sub hi()
-         Dim sector_num() As String
-         sector_num = Split(prompt.Text, vbCrLf)
-         For i = 0 To UBound(sector_num)
-             If Strings.Left(sector_num(i), 11) = "block data:" Then
-                 sector = Strings.Right(sector_num(i), Strings.Len(sector_num(i)) - 11)
-             End If
-         Next i
-     End Sub

- proxmark3> hf mf nested 1
- Usage:          
-  all sectors:  hf mf nested  <card memory> <block number> <key A/B> <key (12 hex symbols)> [t,d]          
-  one sector:   hf mf nested  o <block number> <key A/B> <key (12 hex symbols)>          
-                <target block number> <target key A/B> [t]          
- card memory - 0 - MINI(320 bytes), 1 - 1K, 2 - 2K, 4 - 4K, <other> - 1K          
- t - transfer keys into emulator memory          
- d - write keys to binary file          
-            
-       sample1: hf mf nested 1 0 A FFFFFFFFFFFF           
-       sample2: hf mf nested 1 0 A FFFFFFFFFFFF t           
-       sample3: hf mf nested 1 0 A FFFFFFFFFFFF d           
-       sample4: hf mf nested o 0 A FFFFFFFFFFFF 4 A

iceman1001, 2018-09-02:
Sorry,  this is not a visual basic gui repo.  I suggest you ask you question on the forum,  under the windows client section  [http://www.proxmark.org/forum/viewforum.php?id=14](http://www.proxmark.org/forum/viewforum.php?id=14)

If you have a github repo for your GUI ,  then it will be even easier for people to help you.

zywshuai, 2018-09-02:
http://www.proxmark.org/forum/viewtopic.php?id=5955      Please help

closed 2018-09-02 by iceman1001