iceman1001/proxmark3 #241: Suggestion: new client command rem (remark) - can do PR

DidierStevens, 2018-09-02:
I like to refer to the proxmark3.log file to find back data.
To help me with this, I added a rem (remark) command to the client.
This command echoes its arguments, together with a timestamp, like this:

command: rem Library card
output: 2018-09-02T08:26:02Z remark: Library card

This gives me more context when I search through the proxmark3.log

If you are interested in this new command, I'll do a PR.

[image]

iceman1001, 2018-09-02:
cool,   I can see that it might help out for ppl who parses the log file.

DidierStevens, 2018-09-02:
Cool! Did a PR.

closed 2018-09-02 by iceman1001