iceman1001/proxmark3 #245: Write flashmem while on standalone mode (RDV4)

bogiton, 2018-09-05:
The Flash_WriteData function doesn't seem to be working while on standalone mode with no connected client. I will try to describe my issue.
I was playing around with the new mem commands that work perfectly through the client. I then had a thought to try and write bytes on the flash while sniffing (e.g. just the auth commands). So I did some quick changes to the SniffIso14443a function to achieve just that. In order to see the debug messages, my first step was to try it out while connected to the client. I pressed the button for over 1 second and I got the message that the sniffing started (that was my standalone mode). I then tested sniffing over a reader and the flash memory was successfully written.

Memory before the sniffing session:
```
pm3 --> mem read o 0 l 32
#db# FlashMem read | 0 - 32
#db# FlashMem reading  | 0 | 0 | 32
#db# [Chunk: :0/32] ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
#db# [Chunk: :16/32] ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```
Then I started sniffing by holding the button
```
#db# Starting to sniff
...
#db# maxDataLen=189, Uart.state=0, Uart.len=0
#db# traceLen=473, Uart.output[0]=0000001b
```
Stopped sniffing with button press and viewed the flash mem:
```
pm3 --> mem read o 0 l 32
#db# FlashMem read | 0 - 32
#db# FlashMem reading  | 0 | 0 | 32
#db# [Chunk: :0/32] 59 ab ef f8 ff ff ff ff ff ff ff ff ff ff ff ff
#db# [Chunk: :16/32] ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```
The auth_pwd was successfully written.
I then wiped the flash mem, disconnected the rdv4 and powered it up with a powerbank.
Did exactly the same thing (although no client connected), but nothing got written in the flash mem.
Any idea why this might be happening?

iceman1001, 2018-09-05:
... I suggest you use  the RDV40 dedicated repo.  https://github.com/rfidresearchgroup/proxmark3

There is a few changes coming to flashmem :)  Among others a better driver thanks to @cjbrigato 
Look inside his PR for flashmem use inside a standalone mode.

Your standalone mode seem interesting,  does it have anything to do about a greek transport system ;)
ANyway, I would think your issues here has something to do about the debug printing.

bogiton, 2018-09-05:
Amazing! It was indeed the debug messages that caused the issue.
Works nicely now! Thanks a lot! :)

closed 2018-09-05 by bogiton

cjbrigato, 2018-09-05:
Hello.
Stay tuned for a strong pass2 on my own first modification of the flashmem driver. In the meantime, you can take a look on how I manage recording/recalling the last tag broken in pass1, it works flawlessly.

https://github.com/RfidResearchGroup/proxmark3/blob/8d673fa1bf1d578439057f29de6bb19a7820d0a5/armsrc/Standalone/hf_colin.c#L274

https://github.com/RfidResearchGroup/proxmark3/blob/8d673fa1bf1d578439057f29de6bb19a7820d0a5/armsrc/Standalone/hf_colin.c#L216