iceman1001/proxmark3 #247: init i2c

merlokk, 2018-09-06:
here
https://github.com/iceman1001/proxmark3/blob/master/common/i2c.c#L46
standart i2c init. but...  if you reset CPU in i2c exchange cycle you will have "hanged" i2c. in this situation you can unhang i2c only via power cycle.

here some code to avoid this situation:
https://github.com/merlokk/SmartHome/blob/master/ESP8266CO2PM25/ESP8266CO2PM25.ino#L75
it have written for arduino, but i think the idea is clear)

p.s. i dont dave rdv4 by now so i cant write a PR(

iceman1001, 2018-09-06:
i2c has been transformed,   https://github.com/RfidResearchGroup/proxmark3/blob/master/common/i2c.c

but I still don't understand the reset CPU..  Where do you see a reset CPU?

merlokk, 2018-09-06:
CPU reset - was a sample. generally, any interrupt of i2c transmition may hang i2c bus.
I use that code in situation with many devices on i2c bus and very high load. Some time i2c bus "hangs" and cpu reset dont "un-hang" it.

iceman1001, 2018-09-13:
@merlokk    I thought your company would let you get your hands on a rdv40 :)
send me an email and we can take a discussion

cjbrigato, 2018-09-14:
He's  right on the matter about the CPU reset not anyhow ensuring i2c recovery. 
@merlokk clever way to handle such case, never thought about it. Still I have never faced such hangs in the past which would have been in need of your kind of workaround. 
I have device which have to share a single hardware i2c between capacitive touchscreen and pn532 rfid reader still I've never ran in such hangs.

I'll check 2 things on my side this weekend :
1. Don't we hit a silicon bug anyhow ? At91sam7sXX[X] are endless bags of such bugs. Should check the oatmeal paper covering them, 50% time a workaround is provided of which 25% are acceptable workaround not needing that much patching.
2. If we are tied to resetting the CPU, aren't there multiple reset type for the mcu, if which one may alleviate the hang (thinking about the stm32 family where the same reset type made to ensure we can attach a debugger in any case would indeed ensure i2c is back to normal).

If 1 or 2 is already known tell me, I'll be glad to avoid reading throw these again.

RfidResearchGroup, 2018-09-16:
I have no idea.

iceman1001, 2018-10-27:
@merlokk   Now that you have access to a RDV4,  do you still think it is a problem or?
If problem,  how about a fix?

merlokk, 2018-10-27:
hi) it may be problem...  in some rare cases but....
ill fix it in near future after end fido part.

closed 2019-01-06 by iceman1001