iceman1001/proxmark3 #259: iClass readblk/dump errors when there's a 0xBB

geekman, 2018-10-10:
I'm trying to dump some iClass cards, and I have managed to create dumps with the standard key so far, except for one particular card, which proxmark has difficulty reading completely:

```
[+] saving dump file - 13 blocks read
[+] saved 104 bytes to binary file iclass_tagdump-xxx.bin
#db# [!] block 0D failed to read
```

It seems that block `0xD` fails to read, causing the `dump` command to stop short.

My attempts to use `readblk` to read `0xD` also fail, but I am able to read the remaining blocks from `0xE` onwards till `0x1F`.

I thought it was some kind of RF problem, so I tried using `hf list iclass`, and I saw the same data being returned consistently across the read commands, CRC also said OK, but yet the command "failed":

```
| Src | Data (! denotes parity error)          |CRC | Annotation
+-----+----------------------------------------+----+--------------------
|     |   . . .                                |    |
| Rdr |05  00  00  00  00  54  19  53  08      |    | CHECK
| Tag |9e  8b! 56! 42!                         | ok | 
| Rdr |0c  0d  96  e8                          | ok | READ(13)
| Tag |xx  xx! xx  xx  xx  xx  xx  bb! 46  f9! | ok | 
| Rdr |0c  0d  96  e8                          | ok | READ(13)
| Tag |xx  xx! xx  xx  xx  xx  xx  bb! 46  f9! | ok | 
| Rdr |0c  0d  96  e8                          | ok | READ(13)
| Tag |xx  xx! xx  xx  xx  xx  xx  bb! 46  f9! | ok | 
| Rdr |0c  0d  96  e8                          | ok | READ(13)
| Tag |xx  xx! xx  xx  xx  xx  xx  bb! 46  f9! | ok | 
| Rdr |0c  0d  96  e8                          | ok | READ(13)
```

Digging into the source led me to [`sendCmdGetResponseWithRetries`](https://github.com/iceman1001/proxmark3/blob/master/armsrc/iclass.c#L1835) where it happened to be checking for `0xBB` in `resp[7]`, which is what the traces show.

I am not an expert in the iClass air protocol, but I believe that the card is returning 8 bytes, as expected because each block is 8 bytes. The last 2 bytes are what I am assuming is the CRC. 

In this case, it looks like there is a bug in the error checking logic?

iceman1001, 2018-10-13:
0xBB is a indicator that the iclass demodulation failed.   One of those strange quirks in the source code when debug messages are printed to the trace log with intention.

iceman1001, 2019-09-14:
May I suggest you use RRG/Iceman repo for your device.  https://github.com/rfidresearchgroup/proxmark3

It has gotten some iClass love.

closed 2019-09-14 by iceman1001