iceman1001/proxmark3 #261: lf t55xx detect command cann't detect t5577 card [blocok 0 000480e8]

feifeilanniao, 2018-10-21:
[image]
I try to detect the t5577 card, it runs well when i config the block 0 with data [001C80e8](RF/128). But when i config  block 0 to [000480e8] (RF/16), it can't.

iceman1001, 2018-10-21:
... the iceman fork suffers on LF ,  I haven't moved all fixes from the RRG repo.
https://github.com/RfidResearchGroup/proxmark3
if you use that one,  you will find it works better

iceman1001, 2018-10-27:
You would need to edit the makefiles,  already mentioned in some issues,   and it works fine on other devices.  see  [http://www.proxmark.org/forum/viewtopic.php?pid=33233#p33233](http://www.proxmark.org/forum/viewtopic.php?pid=33233#p33233)

If you have questions on RRG repo you need to ask there.   Flash issues ppl has when swapping between repos has been answered numerous times on both the forum and on GH issues.  Search and you will find the answers.  Start here [https://github.com/RfidResearchGroup/proxmark3/issues/35](https://github.com/RfidResearchGroup/proxmark3/issues/35)

iceman1001, 2019-03-11:
Latest fixes for RRG repo solves this one.

closed 2019-03-11 by iceman1001