iceman1001/proxmark3 #265: Flash memory with rdv4

vdwel, 2018-10-31:
When I load data in the flash memory, for example a binary file containing 4 bytes on offset 0:
00 00 00 00
And load 4 bytes containing: AA BB CC DD EE
The memory keeps at: 00 00 00 00

Further observation has made me come to the conclusion that data written to the flash memory can only change bits from 1 to 0, but not back to 1. So wiping the memory first and then loading the data, does work.

Since I am writing a standalone application for the pm3 rdv4, which I would like to build in such a way that it can remember data (UID's). This is rather inconvenient. I did look at the flash memory code, but couldn't find a fix for this.

Some help would be appreciated.

iceman1001, 2018-10-31:
First of all I would say you are in the wrong repo.
The dedicated repo for RDV4:   https://github.com/RfidResearchGroup/proxmark3

Secondly,  flashmem isn't like ram.  Read this issue to understand better:
https://github.com/RfidResearchGroup/proxmark3/issues/34

vdwel, 2018-10-31:
Thanks!! I'll dive into those.

vdwel, 2018-10-31:
Exactly what I was experiencing. :-) Thanks for pointing out the links.

closed 2018-10-31 by vdwel