iceman1001/proxmark3 #268: failure to link bootrom - section `.interp' will not fit in region `bootphase1'

vyrus001, 2018-11-15:
arm-none-eabi-gcc -nostartfiles -nodefaultlibs -Wl,-gc-sections -n -Wl,-T,ldscript-flash,-Map,obj/bootrom.map -o obj/bootrom.elf obj/version.o obj/ram-reset.o obj/flash-reset.o obj/cmd.o obj/usb_cdc.o obj/bootrom.o -lgcc
/usr/libexec/gcc/arm-none-eabi/ld: obj/bootrom.elf section `.interp' will not fit in region `bootphase1'
/usr/libexec/gcc/arm-none-eabi/ld: section .interp LMA [0000000000100200,0000000000100210] overlaps section .bootphase2 LMA [0000000000100200,000000000010109b]
/usr/libexec/gcc/arm-none-eabi/ld: region `bootphase1' overflowed by 276 bytes
/usr/libexec/gcc/arm-none-eabi/ld: obj/cmd.o: in function `cmd_send':
cmd.c:(.text.cmd_send+0x8e): undefined reference to `__stack_chk_fail'
/usr/libexec/gcc/arm-none-eabi/ld: cmd.c:(.text.cmd_send+0xa8): undefined reference to `__stack_chk_guard'
/usr/libexec/gcc/arm-none-eabi/ld: obj/usb_cdc.o: in function `AT91F_CDC_Enumerate':
usb_cdc.c:(.text.AT91F_CDC_Enumerate+0x1de): undefined reference to `__stack_chk_fail'
/usr/libexec/gcc/arm-none-eabi/ld: usb_cdc.c:(.text.AT91F_CDC_Enumerate+0x2fc): undefined reference to `__stack_chk_guard'
/usr/libexec/gcc/arm-none-eabi/ld: obj/bootrom.o: in function `flash_mode':
bootrom.c:(.text.flash_mode+0x8c): undefined reference to `__stack_chk_guard'
/usr/libexec/gcc/arm-none-eabi/ld: warning: creating a DT_TEXTREL in object
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:43: obj/bootrom.elf] Error 1

vyrus001, 2018-11-25:
gcc version 8.2.0 (Gentoo Hardened 8.2.0-r2 p1.2)

iceman1001, 2018-11-25:
Looks like the new gentoo arm toolchain creates a too large bootrom image.  Stack_chk_guard ..
Under the wiki  https://github.com/Proxmark/proxmark3/wiki/Gentoo-Linux   it states another arm toolchain.  Tried it?

fabled, 2019-01-04:
I had the same issue. It is because gcc generates PIE executables by default. Add -no-pie to CFLAGS to fix it.

iceman1001, 2019-01-05:
@fabled   where do you add this?  the bootrom & armsrc makefile?
**-no-pie**
**-fno-stack-protector**

closed 2019-01-06 by iceman1001