iceman1001/proxmark3 #271: hf mf sim not working reliably

fabled, 2019-01-05:
I just got my proxmark3 rdv4.0 and I am testing mifare side things. It seems the 'hf mf sim' is not currently working reliably. I tested the following:
 - Proxmark official master - 'hf mf sim' works reliably
 - iceman1001 master - 'hf 14a sim' works reliably
 - iceman1001 master - 'hf mf sim' does not work reliably

What happens is that the A-led keeps blinking when close to reader, but the reader is unable to select the emulated token. Occasionally the reader is able to select the token I am emulating.

Perhaps this is due to the fact that 'hf 14a sim' and Proxmark official git master 'hf mf sim' are using prepare_allocated_tag_modulation() + EmSendPrecompiledCmd() for the initial handshake (anticollision + selection) allowing faster responses, where as this branches' 'hf mf sim' is using the plain EmSendCmd(). I suppose this causes timing differences that making it work unreliable against some readers.

iceman1001, 2019-01-05:
aha...
when it comes to RDV4,   I suggest you look in https://github.com/RfidResearchGroup/proxmark3  dedicated repo. 

And yes,  iceman fork hasn't gotten the remake of **sim** from official repo yet.

iceman1001, 2019-04-22:
@fabled  There has been some fixes pushed to the RRG repo last weeks regarding  **hf mf sim**
I will close this issue and you can report back in the RRG repo.    See my prev post for link.

closed 2019-04-22 by iceman1001