iceman1001/proxmark3 #275: Command execute time-out

port9org, 2019-01-25:
Hi guys, 

I'm running the latest fork of iceman's px3 fork (bootloader + os). I try to dump a legic tag with 256 bytes and it works:

pm3 --> hf legic dump
TYPE : MIM256 card (234 bytes)
Reading tag memory 256 b...
Wrote 256 bytes to XXXXXXXX.bin

but if i try to run the same command against a 1024b tag i get a timeout:
pm3 --> hf legic dump
TYPE : MIM1024 card (1002 bytes)
Reading tag memory 1024 b...
Command execute time-out

same for "hf legic info".
Works for the 256 card but not for the 1024 card:

----
pm3 --> hw ver

Proxmark3 RFID instrument

 [ CLIENT ]
 client: iceman build for RDV40 with flashmem; smartcard;

 [ ARM ]
 bootrom: iceman/master/ice_v3.1.0-1070-ge622da56 2019-01-25 16:14:40
      os: iceman/master/ice_v3.1.0-1070-ge622da56 2019-01-25 16:14:42

 [ FPGA ]
 LF image built for 2s30vq100 on 2017/10/25 at 19:50:50
 HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23

 [ Hardware ]
  --= uC: AT91SAM7S256 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 256K bytes, Used: 237350 bytes (91%) Free: 24794 bytes ( 9%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

pm3 --> hw tune

[=] measuring antenna characteristics, please wait...

....

[+] LF antenna: 25.35 V - 125.00 kHz
[+] LF antenna: 18.98 V - 134.00 kHz
[+] LF optimal: 25.35 V - 122.45 kHz
[+] LF antenna is OK

[+] HF antenna: 19.35 V - 13.56 MHz
[+] HF antenna is OK

[+] Displaying LF tuning graph. Divisor 89 is 134khz, 95 is 125khz.

iceman1001, 2019-01-25:
this repo hasn't gotten much love for awhile.   There were some timout issues addressed in the RRG repo,  you can look at them and apply it to your code.   Most likely same issues.

port9org, 2019-01-25:
HI @iceman1001  thanks for your reply! I'll check.
you are referring to issue 83 https://github.com/RfidResearchGroup/proxmark3/issues/83 ?

just to clarify, is the repo RfidResearchGroup/proxmark3 *only* for proxmark rdv4.0 or can I use this with my px3 rdv2 easy aswell?

thanks!

iceman1001, 2019-01-25:
Yes,   issue 83. 

Regarding older device and RRG repo,  I wrote a post about it.   Its quite easy to compile for non rdv4.
http://www.proxmark.org/forum/viewtopic.php?pid=33233#p33233

closed 2019-01-25 by iceman1001