iceman1001/proxmark3 #277: hf 14a sim t 9 not work

SpenserCai, 2019-02-11:
if use 
hf 14a sim t 9
will back unknow tagtype(9)
but
hf 14a help have information about this
t: 1=...
   2=...
   ......
   9=FM11RF00SH ShangHai Metro

iceman1001, 2019-02-13:
I don't think that mode is finshed.  There were a chinese dude (I forgotten whom, sorry) working with it but than I have heard nothing more from it. 

If you want to finish it?
or do you have access to such cards/readers?

iceman1001, 2019-02-13:
And have you tried the  https://github.com/RfidResearchGroup/proxmark3 repo?  It has more fixes in the code since I have had time with this repo for awhile.

SpenserCai, 2019-02-14:
I sniffed some information and found that the data for auth was not completely random

pm3 --> hf list 14a
trace pointer not allocated
Recorded Activity (TraceLen = 614 bytes)

Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
iso14443a - All times are in carrier periods (1/13.56Mhz)

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |        448 | Tag |05!                                                                      |     |
    1342576 |    1344944 | Tag |03  00                                                                   |     |
    1512928 |    1515296 | Tag |03  00                                                                   |     |
    1679456 |    1681824 | Tag |03  00                                                                   |     |
    1837632 |    1838848 | Tag |00                                                                       |     |
    1877248 |    1880320 | Tag |4d  8d! 03!                                                              |     |
    2223484 |    2224476 | Rdr |52                                                                       |     | WUPA
    2225744 |    2228112 | Tag |03  00                                                                   |     |
    2256316 |    2261084 | Rdr |30  01  8b  b9                                                           |  ok | READBLOCK(1)
    2262272 |    2269248 | Tag |b7  b4  68  e0  05  c5                                                   |  ok |
    2275772 |    2280540 | Rdr |30  00  02  a8                                                           |  ok | READBLOCK(0)
    2281728 |    2288704 | Tag |03  00  02  90  f4  d4                                                   |  ok |
    2307516 |    2312284 | Rdr |30  01  8b  b9                                                           |  ok | READBLOCK(1)
    2313472 |    2320448 | Tag |b7  b4  68  e0  05  c5                                                   |  ok |
    2326204 |    2330908 | Rdr |30  02  10  8b                                                           |  ok | READBLOCK(2)
    2332160 |    2339136 | Tag |00  02  24  db  b5  cf                                                   |  ok |
    2344636 |    2349340 | Rdr |30  03  99  9a                                                           |  ok | READBLOCK(3)
    2350592 |    2357568 | Tag |7b  7c  78  01  a6  c6                                                   |  ok |
    2369024 |    2376000 | Tag |84  3c  06  98  9e  66                                                   |  ok |
    2387456 |    2394496 | Tag |02  42  01  64  42  e0                                                   |  ok |
    2405760 |    2412800 | Tag |5c  63  0a  e2  67  c7                                                   |  ok |
    2423680 |    2430720 | Tag |01  2c  30  30  02  6b                                                   |  ok |
    2610732 |    2615500 | Rdr |60  03  6e  49                                                           |  ok | AUTH-A(3)
    2617088 |    2621824 | Tag |c0  d1  06  89                                                           |     |
    2625196 |    2634508 | Rdr |61  17  0f  93! a9! d3  cb! 25!                                          | !crc| AUTH-B(23)
    2635776 |    2640448 | Tag |78  e5  fa  d3!                                                          |     |
    2645036 |    2649804 | Rdr |e9  56  67! c9!                                                          | !crc|
    2651200 |    2652224 | Tag |01                                                                       |     |
    2663084 |    2667788 | Rdr |16! f6  46  fd!                                                          | !crc|
    2669056 |    2676096 | Tag |43  df! b0  37! d0  af!                                                  | !crc|
    2681004 |    2685708 | Rdr |ef  a0! cb  cb!                                                          | !crc|
    2686976 |    2694016 | Tag |05  aa  fe! f4! 2a! ec                                                   | !crc|
    2699436 |    2704204 | Rdr |cb  f7  11  d5                                                           | !crc|
    2705664 |    2712384 | Tag |2e! c1  bf! 10! 9b! 77!                                                  | !crc|
    2717740 |    2722444 | Rdr |c3! d2  88! 2e                                                           | !crc|
    2723712 |    2730688 | Tag |7f  9c! 89  e8! 13! a2!                                                  | !crc|
    2736300 |    2741004 | Rdr |47! ed! 10! be                                                           | !crc|
    2742400 |    2749312 | Tag |f6  4f! 4b  ff! bd! 1b!                                                  | !crc|
    3753884 |    3758652 | Rdr |eb  a9  b4  3c!                                                          | !crc|
    3760032 |    3760224 | Tag |01                                                                       |     |
    3765276 |    3772348 | Rdr |8c! b3  12! d6! 54  b5                                                   | !crc|
    3814864 |    3815248 | Tag |01                                                                       |     |
    3820316 |    3825084 | Rdr |7f! a4  3b  97                                                           | !crc|
    3831836 |    3838908 | Rdr |29  76  f3  f2  1d  a4                                                   | !crc|
    3881168 |    3881808 | Tag |03!                                                                      |     |
    3886748 |    3891516 | Rdr |b5  b6! c3! 9e                                                           | !crc|
    3893136 |    3893328 | Tag |01                                                                       |     |

SpenserCai, 2019-02-14:
pm3 --> hf 14a raw -s -c 6001
7B 37 F1 D5
pm3 --> hf 14a raw -s -c 6001
64 E5 BA D7
E5 BA D7 1E
54 C9 61 8C
E5 BA D7 1E
BA D7 1E 4E
35 1B 47 06
64 E5 BA D7
E5 BA D7 1E
47 06 2C E7
E5 BA D7 1E
D7 1E 4E 2A
64 E5 BA D7
BA D7 1E 4E
5A 64 E5 BA
BA D7 1E 4E
E5 BA D7 1E
BA D7 1E 4E
E5 BA D7 1E
BA D7 1E 4E
5A 64 E5 BA
63 5A 64 E5
1E 4E 2A C6
D7 1E 4E 2A
E5 BA D7 1E
BA D7 1E 4E
61 8C 96 4A
1E 4E 2A C6
D7 1E 4E 2A
D7 1E 4E 2A
E5 BA D7 1E
64 E5 BA D7
BA D7 1E 4E
2A C6 54 C9
D7 1E 4E 2A
E5 BA D7 1E
BA D7 1E 4E
D7 1E 4E 2A
D7 1E 4E 2A

I think it has some rules like this

5A64E5BAD71EAE2A
351B47062CE7

iceman1001, 2019-02-14:
How about you start a thread over at the proxmark forum?   This is not the place to discuss it.

iceman1001, 2019-02-14:
http://www.proxmark.org/forum/viewtopic.php?pid=34282#p34282

iceman1001, 2019-04-22:
closed because of inactivity

closed 2019-04-22 by iceman1001