iceman1001/proxmark3 #280: Standalone mode enhancements

Mera-balou, 2019-03-01:
It seems that the actual standalone mode only works for unprotected LF cards.
IOT use PM3 RDV 4 on the field, it is possible to enhance standalone mode for automating cracking process without PM3 user interaction?
I mean, if my target is a myfare classic (but I don't know this information of course), standalone mode might be (it's an example):
- lf search during 10 seconds
=> No card found (in this case)
- switch to hf search during 10 s
=> Myfare identified
    - try default/leaked keys
 => No key found
   - darkside attack
  => One key found
   - nested attack
   => All keys found
  => Attack succeed

The process would be adapted according to the device identified (iClass, Ultralight EV1...).
Led state (blinking, remain on) will change depending steps of the attack:
- attack in progress
-  attack succeed
- impossible attack (eg. mifare DESFire)

PS : happy #Nullcon!

iceman1001, 2019-03-01:
of course you can code a standalone mode for that.  Feel free to create it!  It would be a great standalone mode to have.

iceman1001, 2019-09-16:
rethinking this one... no,  there is no solution for running darkside / nested on the device.  It is too slow.

closed 2019-09-16 by iceman1001