iceman1001/proxmark3 #287: Is this a bug?

9844201, 2019-05-05:
[image]
[image]

no display key
===========================================================
but use https://github.com/proxmark3/ 
display key

[image]

iceman1001, 2019-05-05:
depends.
looks like you have mixed firmware & client from  iceman repo vis official repo.  
The general recommendation is never to mix firmware & client from different builds, repos.  Always use the same client as you used when you flashed.

9844201, 2019-05-05:
[image]
The firmware and client are the same

9844201, 2019-05-05:
[image]
Firmware and client are the same version

iceman1001, 2019-05-05:
if you look at the found keys in your run with official repo,   00270088bf51 it isn't in the default keys used in your iceman run.   The res column indicates success (1)/ failure (0)

The bug would be why official repo finds a key not used in the dictionary.

closed 2019-05-05 by iceman1001