iceman1001/proxmark3 #292: ubuntu 18.3 install instructrions in README.md

EtoTen, 2019-06-13:
I was getting a build error doing the build after following the instructions in README.md  on Ubuntu 16.04.

"...
arm-none-eabi-gcc -c -I../include -I../common -I. -Wall -Werror -pedantic -Wunused -std=c99 -I. -fno-strict-aliasing -ffunction-sections -fdata-sections -Os -mthumb -mthumb-interwork -o obj/version.o version.c 
In file included from ../include/usb_cmd.h:22:0,
                 from ../include/proxmark3.h:17,
                 from version.c:1:
/usr/lib/gcc/arm-none-eabi/4.9.3/include/stdint.h:9:26: fatal error: stdint.h: No such file or directory
 # include_next <stdint.h>
...
"
Installing an additional library package via "sudo apt-get install libnewlib-arm-none-eabi" resolved it, everything built and the flasher and client worked.

Can this additional package install requirement be added to the README.md?

iceman1001, 2019-06-13:
So you are saying gcc-arm-none-eabi didn't work for you on   Ubuntu 18.3 or Ubuntu 16.04 ?

```
sudo apt-get install p7zip git build-essential libreadline5 libreadline-dev libusb-0.1-4 libusb-dev libqt4-dev perl pkg-config wget libncurses5-dev gcc-arm-none-eabi
```

EtoTen, 2019-06-13:
Sorry, the 18.3 in the title was a typo. It is Mint 18.3 which is Ubuntu 16.04.

I copy and pasted the line from "Run sudo apt-get install p7zip git build-essential libreadline5 libreadline-dev libusb-0.1-4 libusb-dev libqt4-dev perl pkg-config wget libncurses5-dev gcc-arm-none-eabi" 

but still received the error. 

So I guess gcc-arm-none-eabi  was not enough.

https://packages.ubuntu.com/xenial/libs/libnewlib-arm-none-eabi

file list:
https://packages.ubuntu.com/xenial/all/libnewlib-arm-none-eabi/filelist

https://packages.ubuntu.com/xenial/gcc-arm-none-eabi

file list:
https://packages.ubuntu.com/xenial/amd64/gcc-arm-none-eabi/filelist

iceman1001, 2019-08-15:
This fork is deprecated,  use  RRG/Iceman repo instead. ( https://github.com/rfidresearchgroup/proxmark3 )

closed 2019-08-15 by iceman1001