iceman1001/proxmark3 #293: Hf mf chk problem

xianglin1998, 2019-06-27:
hello, mr'iceman，I am dxl，When i using you open source "proxmark3" ，I found a problem，I don't know if it was my fault，In the hf mf chk, I can't execute，Tracking source code, i found:
In the function "CmdHF14AMfChk"
`//code
//        case '?':
//                keyType = 2;
// defined
 uint8_t trgKeyType = 0;
.......
 for (trgKeyType  = keyType ; trgKeyType < 2; (keyType == 2) ? (++trgKeyType) : (trgKeyType = 2)) 
// print in the param "?" keyType = 2 
look, you set the trgKeyType = 2 in the first, the loop can't access`

xianglin1998, 2019-06-27:
fix:  for (; trgKeyType < 2; (keyType == 2) ? (++trgKeyType) : (trgKeyType = 2))

xianglin1998, 2019-06-27:
remove the first value set, is can work. :)

iceman1001, 2019-06-27:
Your name showed up in revE rebooted firmware createbin.exe etc :)

I strongly suggest you use  https://github.com/RfidResearchGroup/proxmark3  repo
it has more fixes,  and can be compiled for other proxmark3 devices than rdv4.

There is a bug definitly.
its two paths,  either select a target keytype (A or B)   or  any keytype (?) (check both)
your suggestion removes the first path,  how to set target keytype A or B, more specificly when you set it to B (1)...

xianglin1998, 2019-06-27:
hhh, I also found this problem. so i change to:
uint8_t trgKeyType = keyType < 2 ? keyType : 0;
if than of 2，using the key A...

iceman1001, 2019-06-27:
make PR?   both here and on RfidResearchGroup repo?

closed 2020-03-01 by iceman1001