iceman1001/proxmark3 #294: Compilation problem

nemanjan00, 2019-07-07:
```
arm-none-eabi-gcc -c -I../include -I../common -I. -Wall -Werror -pedantic -Wunused -std=c99 -DWITH_CRC -DON_DEVICE -DWITH_LF -DWITH_HITAG -DWITH_ISO15693 -DWITH_LEGICRF -DWITH_ISO14443b -DWITH_ISO14443a -DWITH_ICLASS -DWITH_FELICA -DWITH_FLASH -DWITH_SMARTCARD -DWITH_HFSNOOP -DWITH_LF_SAMYRUN -fno-strict-aliasing -ffunction-sections -fdata-sections -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED -I../zlib -I. -Os -mthumb -mthumb-interwork -o obj/appmain.o appmain.c 
appmain.c: In function 'UsbPacketReceived':
appmain.c:622:40: error: taking address of packed member of 'struct <anonymous>' may result in an unaligned pointer value [-Werror=address-of-packed-member]
  622 |    setSamplingConfig((sample_config *) c->d.asBytes);
      |                                        ^
cc1: all warnings being treated as errors
make[1]: *** [../common/Makefile.common:77: obj/appmain.o] Error 1
make[1]: Leaving directory '/home/nemanjan00/gitProgs/proxmark3/armsrc'
make: *** [Makefile:35: armsrc/all] Error 2
```

Temporary solution is removing -Werror from common/Makefile.common... I am not C/C++ guy, so, was not able to solve real problem.

iceman1001, 2019-07-07:
@doegox  This one is for you,  Arch linux...     I suppose this exists in RRG repo aswell. 
Unaligned pointer access.

iceman1001, 2019-07-07:
@nemanjan00  you wouldn't mind testing  https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Linux-Installation-Instructions.md   and compiling the RRG repo?

nemanjan00, 2019-07-07:
Compiles no problems

doegox, 2019-07-08:
Yes I tested rrg on Arch docker, compiles fine

iceman1001, 2019-07-08:
sounds like I can close this one.

closed 2019-07-08 by iceman1001

nemanjan00, 2019-07-08:
RRG does compile, but, this one does not

iceman1001, 2019-07-08:
use RRG,  adjust Makefile.platform file to compile for your proxmark3 device

nemanjan00, 2019-07-08:
Ok, thanks!